ARM-software/lisa #15: [RTA] Could we better integrate RTA workload generator with TestEnv/Devlib?

derkling, 2016-01-25:
Reported by: Morten Rasmussen

In the fifth cell of the `simple_rtapp.ipynb`` notebook, why do we have to explicitly set `run_dir=target.working_directory` when we haven't changed it (still default)?

The `RTA.conf()` method should be able to retrieve the information from the target or `TestEnv.target` object if it isn't explicitly specified.

derkling, 2016-01-25:
I need a check if I can properly initialize the default value in case we do not pass a run_dir explicitely.

valschneider, 2018-09-05:
This `run_dir` has proven to be a pain point, especially for newcomers to LISA. I've hidden it away entirely in my wlgen rework (https://github.com/valschneider/lisa/tree/tests/executor-rework).

closed 2018-09-05 by valschneider