ARM-software/lisa #23: [TestEnv] Setup devlib modules based on board definition

derkling, 2016-01-25:
For the set of known boards, we know which modules can be loaded. For example "hwmon" is quite
likely required only by "TC2" and "Juno".

We could exploit the board definition at TestEnv initialization time to define the list of modules to initialize when the devlib target is build.

This should allow to simplify further the default configuration file as well as make some test more portable across different boards.

derkling, 2016-01-27:
We should still consider that for a spefific platforms sometimes we can we interested in "not loading" a specific module. For example, if we are not interested to energy measurements, we would like to not load the "hwmon" module. This should result in faster target setup times.

However, perhaps unnecessary modules loading can still be tolerated provided that all modules are quite fast in probing and loading. In that case the load of all modules supported by a platform should not be a big issue. Thus, perhaps that patch should be done once we fixed at least #5

deggeman, 2016-01-28:
That would for instance be helpful to run rfc eas tests on Hikey board. 
Right now I have to comment out the "bl" module in target.config and the  "hwmon" and "bl" module in tests/eas/rfc_eas.config. The first one would be OK but the need to adapt the test config file should be avoided.

derkling, 2016-01-29:
The solution I'm thinking about is:

- for each known board we define the **complete** list of modules supported by that board in board specific section of [libs/utils/env.py](https://github.com/ARM-software/lisa/blob/master/libs/utils/env.py#L285), e.g.
```python
    # Initialize a specific board (if known)
    if self.conf['board'].upper() == 'TC2':
        platform = devlib.platform.arm.TC2()
        self.__modules = ['bl', 'hwmon']
```
- in the main target.config file we expose a new "exclude modules" which allows to mask off some of the board specified modules. This should allow for example to avoid loading "hwmon" in TC2 if we do not need that module for our own experiments, e.g.
```python

    /* Devlib modules to enable/disable for all the experiments */
    "modules"  : [ ],
    "exclude_modules" : [ "hwmon" ],

derkling, 2016-01-29:
Possible fix in pull reqeust #31

commit 367369e5 (2016-01-29): Merge pull request #31 from derkling/update_modules_initialziation

Fix #23
closed 2016-01-29 by deggeman