ARM-software/lisa #6: [TraceAnalysis] Return dictionary of collected metrics

derkling, 2016-01-25:
While the `libs/utils/trace_analysis.py` module parses the output trace it generates a set of interesting metrics which are reported only as debugging logs.
It could be useful in general to build a dictionary collecting all these metrics and to make it available to external code.

mdigiorgio, 2016-09-16:
Is this relevant anymore? `libs/utils/trace_analysis.py` has been removed...

derkling, 2016-09-19:
Yes, I agree this is now kind-of obsolete... we still dump some information while the trace is parsed but I think we can already access these info (e.g. over-utilized time percentage) on demand after the trace has been parsed. Let's close this for the time being.

closed 2016-09-19 by derkling