ARM-software/lisa #49: Is BART's commit ID wrong?

Leo-Yan, 2016-02-11:
Hi,

I found after I run command "source init_env", then it will checkout BART to commit id b7fe442093cd3ea4c2148f1e56c9b0b793238654; but when I check bart's source code with this commit ID, there have no real python lib. Please check below BART's tree:

https://github.com/ARM-software/bart/tree/gh-pages.

Thanks,
Leo Yan

derkling, 2016-02-11:
On 11-Feb 07:25, Leo Yan wrote:
> Hi,

Hi Leo,
 
> I found after I run command "source init_env", then it will checkout
> BART to commit id b7fe442093cd3ea4c2148f1e56c9b0b793238654; but when
> I check bart's source code with this commit ID, there have [...]

You are right, by error I was pointing to a wrong version of BART.
We have just merged on master a patch which should fix that issue.
The really same patch:
  https://github.com/ARM-software/lisa/commit/d510d48af521db2b7da530cf3d0694e935424b5d
changes also the remote we are tracking from within LISA.

It would be great if you can pull the current master and try again
with "source init_env". Hopefully submodules should update and
checkout the proper version.

After that command you can check which version you have with this
commad:
  $ git submodule status

You should get that outout:
    290caafb3675103f8ee91b71245a7aba0f2b0c2c libs/bart (v1.5.0)
    6586d8b10454fad4299ded9f9f31058fa6d66549 libs/devlib (v0.0.3-38-g6586d8b)
    26cf414f6b6c5eca30ef12822bc03e06cb1adc34 libs/trappy (v5.1.0)

Lemme know if there are issues.

Cheers Patrick

-- 
#include <best/regards.h>

Patrick Bellasi

derkling, 2016-02-11:
We just tested here again with another machine and fixed a couple of more issues.
Patches are already in mainline, however you probably need to
```bash
$ rm -rf libs/bart
$ rm -rf libs/trappy
```
before doing a "source init_env", otherwise apparently the new remote is not used to pull updates for that paths.

That should be hopefully enough to get a working checkout from the new remote. However, if you have a look at your _.git/config_ you should still find some submodules named for "bart-next" and "trappy-next".
These two entries needs to be removed by hand.

Leo-Yan, 2016-02-12:
Hi Patrick,

> We just tested here again with another machine and fixed a couple of more issues.
> Patches are already in mainline, however you probably need to
> ```bash
> $ rm -rf libs/bart
> $ rm -rf libs/trappy
> ``` [...]

Have verfied, now it can work well at my side. Thanks for fixing.

Thanks,
Leo Yan

derkling, 2016-02-12:
Unfortunately we verified from our side as well that we need to remove by hand the old libraries as well as update the .git/config to remove "*-next" submodules. That's a bit of a pain but hopefully it should not be so common to update the tracked remote repositories.
In the future, we will probably add a more robust update procedure which automate that manual steps.

Thanks for testing, I'm going to close that issue in the following days if there will not be other complains.

Cheers Patrick

JaviMerino, 2016-03-31:
I think this is solved now

closed 2016-03-31 by JaviMerino