ARM-software/lisa #84: Results processor for LISA

raw-bin, 2016-04-04:
[Disclaimer: The enhancements here may not be applicable to LISA alone but I'll mention them here all the same to gather viewpoints. I've also not been through all the issues in the issue list - so there may be some redundancy].

There's a set of common 'views' of data that can be very useful in diagnosing a given use-case or system to form a first order viewpoint for diagnosis etc. 

For example - in the past cr2 produced a fixed set of plots that were very valuable to form an initial opinion of what's up. It would be good to have a similar setup with one 'report' generator that first confirms that all necessary tokens are available in input trace and if so, produces all the views.

It's kind of easy to get carried away with this sort of stuff admittedly - one type of view can sometimes be inferred from another so you don't always want to plot every type of view. However, in my experience, if the generation cost is minimal, having multiple views of interest - some redundancy notwithstanding - is a very useful correlation tool to diagnose problems.

Some of the views that I can think of are:

1. OPP residency break-down for a given trace input.
  * Per-CPU OPP residency break-down in a suitable plot.
  * Great to know where the compute's going. For example, Audio playback should show low OPPs on the LITTLE side only etc.
  * @JaviMerino has an implementation internally which might be reusable.

2. Idle state residency break-down.
  * Per-CPU/cluster idle residency break-down in a suitable plot.
  * Similar to 1 above, can help correlate problems. 

3. Parallelisation scoring for a given trace input.
  * Aggregate scoring - ie what %age of time were N cpus active (N: {0,1,...NCPUs}) - in a suitable plot.
  * Most obvious use is for benchmarks where you want to verify for example 1 thread per CPU kind of dynamic.
  * WA has a [result processor](https://github.com/ARM-software/workload-automation/blob/master/wlauto/result_processors/cpustate.py) that does this and the above. IMHO those kind of processors are better implemented in LISA.

4. IRQ histogram by CPU
  * Great for confirming who's causing frequent wake-ups on the wrong CPU type for example.

derkling, 2016-04-28:
The first bullet has been probably covered by Michele in PR #87 which is still under review.
Need to send some feedbacks and than we are ready for merge.

mdigiorgio, 2016-05-02:
PR #87 provides per-cluster frequency residency. Using the same approach we can implement another function to get per-CPU data.

Again, similar operations should allow us to produce point 2.

mdigiorgio, 2016-06-24:
Point 1 is available as per #132

douglas-raillard-arm, 2024-02-09:
Closing as this is now ~8 years old and LISA has a number of analysis functions, possibly overlapping with that.

closed 2024-02-09 by douglas-raillard-arm