ARM-software/lisa #85: Lisa usage  issue  for  A  new  board.

zuozhengkun, 2016-04-12:
Hello,
  Now i  kick off new  board,  it  is  android device,  ( target.config )
1.  How to define and configure this  devices,  such as  adb  devices communication.

2. config items:

 /* Binary tools to install by default for all experiments */
    "tools"    : [],    # what  to  be filled.

"rtapp-calib" :  {
    //     "0": 354, "1": 138, "2": 138, "3": 363, "4": 355, "5": 357
    // },
rtapp-calib: how to  get these cores calib data by  which  ways...?

Thanks.

mdigiorgio, 2016-04-12:
`tools` depends on what you want to do:
 - you want to run an `rt-app` experiment -> put `"tools" : ['rt-app']`
 - you want to collect traces using `ftrace` -> put `"tools" : ['trace-cmd']`

For what concerns `rtapp-calib`, if you comment that entry, calibration will be done automatically by LISA and you will find the values for your target in the output of `TestEnv`. You can then copy and paste the values you obtain in your `target.config` to avoid recomputing them every time you run the experiment. Anyway, this is also written in the comments in the default `target.config`, use that as a reference for the different options.

zuozhengkun, 2016-04-12:
OK.

zuozhengkun, 2016-04-13:
Hello,
  Maybe  the  work   environment  is  already  setup  OK.
and  i   modified  the  file   target.config (related  to  my  android device  phone ), only  as followings:

{
    /* Platform */
    /* - linux   : accessed via SSH connection                    */
    /* - android : accessed via ADB connection                    */
    /* - host    : run on the local host                          */
    "platform" : "android",

    /* Board */
    /* Currently supported boards are:                            */
    /*  juno : target is a JUNO board                             */
    /*  tc2  : target is a TC2 board                              */
    /* Leave commented if your board is not listed above          */
    "board" : "juno",   // ???

    /* Target IP or MAC address */
   // "host" : "192.168.0.20",

    /* Target Android device ID */
  //  "device" : "SP9838A210057260251",

    /* Login username (has to be sudo enabled) */
    "username" : "root",

    /* Login credentials */
    /* You can specify either a password or keyfile */
//    "password" : "",
    // "keyfile"   : "/complete/path/of/your/keyfile",

    /* RT-App Calibration */
    /* Keep commented to do calibration at first run on a target. */
    /* Then, replace the values with the ones reported for your   */
    /* specific target in the console logging messages            */
    // "rtapp-calib" :  {
    //     "0": 354, "1": 138, "2": 138, "3": 363, "4": 355, "5": 357
    // },

    /* FTFP Image server */
    /* This is the folder from where the target gets kernel/DTB   */
    /* images at each boot.                                       */
    /* The path of images to be deployed are specified by the     */
    /* experiments configuration (e.g. tests/eas/rfc_eas.json)    */
   /* "tftp"  : {
        "folder"    : "/var/lib/tftpboot",
        "kernel"    : "kern.bin",
        "dtb"       : "dtb.bin"
    }, */

    /* Devlib modules to enable/disbale for all the experiments */
    "modules"         : ["cpufreq",],
    "exclude_modules" : [],

    /* Binary tools to install by default for all experiments */
    "tools"    : ["rt-app","trace-cmd"],

    /* Wait time before trying to access the target after reboot */
    // "ping_time" : "15",

    /* Maximum time to wait after rebooting the target */
    // "reboot_time" : "120",

    /* List of test environment features to enable */
    /*  no-kernel : do not deploy kernel/dtb images               */
    /*  no-reboot : do not force reboot the target at each        */
    /*              configuration change                          */
    /*  debug     : enable debugging messages                     */
    "__features__" : "no-kernel no-reboot"
}
-------------------------------
1.  Are there  the  above  config  parameters wrong? or  right ?

and  because of  my  specified android phone,  and   shell,
lisa-ipython start

2. And  then how  to  run  my default  file  ipynb   under DIR  ipynb/xxx  ?
    which   .ipynb   is  OK  and  configured  well  to  verify  the  first  test  running...

Thanks.

zuozhengkun, 2016-04-13:
2. And then how to run my default file ipynb under DIR lisa/ipynb/xxx ? 
which .ipynb is OK and configured well to verify the first test running...

thanks.

zuozhengkun, 2016-04-14:
Issue 1,
[LISAShell lisa] \> lisa-test  tests/eas/rfc.py

 Failure: ImportError (No module named rfc) ... ERROR

 ======================================================================
 ERROR: Failure: ImportError (No module named rfc)
 ----------------------------------------------------------------------
 Traceback (most recent call last):
   File "/usr/lib/python2.7/dist-packages/nose/loader.py", line 390, in loadTestsFromName
     addr.filename, addr.module)
   File "/usr/lib/python2.7/dist-packages/nose/importer.py", line 39, in importFromPath
     return self.importFromDir(dir_path, fqname)
   File "/usr/lib/python2.7/dist-packages/nose/importer.py", line 71, in importFromDir
     fh, filename, desc = find_module(part, path)
 ImportError: No module named rfc

 ----------------------------------------------------------------------
 Ran 1 test in 0.000s

 FAILED (errors=1)

Issue 2 ,

 [LISAShell lisa] \> lisa-report

 Traceback (most recent call last):
   File "./tools/report.py", line 22, in <module>
     from utils.report import Report
   File "./libs/utils/__init__.py", line 20, in <module>
     from env import TestEnv
   File "./libs/utils/env.py", line 35, in <module>
     from trappy.stats.Topology import Topology
   File "/home/likewise-open/SPREADTRUM/zhengkun.zuo/Documents/EAS/lisa/libs/trappy/trappy/__init__.py", line 24, in <module>
     from trappy.plotter.LinePlot import LinePlot
   File "/home/likewise-open/SPREADTRUM/zhengkun.zuo/Documents/EAS/lisa/libs/trappy/trappy/plotter/__init__.py", line 20, in <module>
     import LinePlot
   File "/home/likewise-open/SPREADTRUM/zhengkun.zuo/Documents/EAS/lisa/libs/trappy/trappy/plotter/LinePlot.py", line 21, in <module>
     from trappy.plotter import AttrConf
   File "/home/likewise-open/SPREADTRUM/zhengkun.zuo/Documents/EAS/lisa/libs/trappy/trappy/plotter/AttrConf.py", line 100, in <module>
     if StrictVersion(matplotlib.__version__) < StrictVersion("1.5.1"):
   File "/usr/lib/python2.7/distutils/version.py", line 40, in __init__
     self.parse(vstring)
   File "/usr/lib/python2.7/distutils/version.py", line 107, in parse
     raise ValueError, "invalid version number '%s'" % vstring
 ValueError: invalid version number '1.1.1rc'

 

In  addition,  my  phone  is  Android device mobile phone by adb  , NOT  network  ssh connected with host .
how  to  config  some files  for  my  first  running.
---> file in  testenv_example.ipynb .. 
   target.config  etc.

Thank  you  very much.

zuozhengkun, 2016-04-14:
OK,
Issue 1 ,  i  found  the cause,  and  solved.

But 
Issue 2 ,
[LISAShell lisa] > lisa-report
Traceback (most recent call last):
 File "./tools/report.py", line 22, in 
from utils.report import Report
 File "./libs/utils/init.py", line 20, in 
from env import TestEnv
 File "./libs/utils/env.py", line 35, in 
from trappy.stats.Topology import Topology
 File "/home/likewise-open/SPREADTRUM/zhengkun.zuo/Documents/EAS/lisa/libs/trappy/trappy/init.py", line 24, in 
from trappy.plotter.LinePlot import LinePlot
 File "/home/likewise-open/SPREADTRUM/zhengkun.zuo/Documents/EAS/lisa/libs/trappy/trappy/plotter/init.py", line 20, in 
import LinePlot
 File "/home/likewise-open/SPREADTRUM/zhengkun.zuo/Documents/EAS/lisa/libs/trappy/trappy/plotter/LinePlot.py", line 21, in 
from trappy.plotter import AttrConf
 File "/home/likewise-open/SPREADTRUM/zhengkun.zuo/Documents/EAS/lisa/libs/trappy/trappy/plotter/AttrConf.py", line 100, in 
if StrictVersion(matplotlib.version) < StrictVersion("1.5.1"):
 File "/usr/lib/python2.7/distutils/version.py", line 40, in init
 self.parse(vstring)
 File "/usr/lib/python2.7/distutils/version.py", line 107, in parse
 raise ValueError, "invalid version number '%s'" % vstring
 ValueError: invalid version number '1.1.1rc'

In addition, my phone is Android device mobile phone by adb , NOT network ssh connected with host .
 how to config some files for my first running.
 ---> file in testenv_example.ipynb .. 
target.config etc.

Thank you very much.

zuozhengkun, 2016-04-14:
Issue 2:  testenv_example.ipynb 
It  is  related  to  your  juno board  by  SSH  network connection, and  6 cpu cores..
But  my  android  phone  is   8 cores. ...
So  how  to customize  and  modify   my   An-phone  with  cpu  cores  and  connections  in  target.config  and  rfc.config.

thanks.

zuozhengkun, 2016-04-15:
Hello,
  According  to  only  Issue 2,   in   file  testenv_example.ipynb 
   Plz   help   to   descript  and clarify   some  key and  useful   items  which  can  be  modified  and  adapt  to  my  android  phone  in  custom  style.  So  it  can   get   some   valid  results  from  testenv.
and  then  reuse  these  parameters ( cal - values ) in  other  tests , such as  rfc.config( rfc.py) .
( eg. my An-phone with 8  cpu  cores  )

Or   are  there  some  document  Actual explaination for  file  testenv_example.ipynb 

Thanks.

mdigiorgio, 2016-06-10:
If your target is an Android phone you should comment out `board` and `host` and specify the device ID under `device` if you have multiple Android devices connected to your host machine.

For example, you can refer to [this bnotebook](https://github.com/ARM-software/lisa/blob/master/ipynb/android/Android_Workloads.ipynb) and all the Android-realted notebooks to understand what parameters we use to configure the target. In particular, you should look at `my_conf` and write a similar `target.config`.

closed 2016-06-14 by zuozhengkun