ARM-software/lisa #89: Run   issues   in  the end    for   help.

zuozhengkun, 2016-04-15:
Hello,
  When  run  finished,  it  produced  some  error  info as fllowings:

Check that there is not regression on energy ... ok
 Check that there is not regression on performance ... ok
 ----------------------------------------------------------------------
 Ran 2 tests in 590.728s

 OK
 Error in atexit._run_exitfuncs:
 Traceback (most recent call last):
   File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
     func(*targs, **kargs)
   File "/usr/lib/python2.7/logging/__init__.py", line 1640, in shutdown
     h.close()
   File "/usr/lib/python2.7/logging/handlers.py", line 1110, in close
     logging.Handler.close(self)
 TypeError: unbound method close() must be called with Handler instance as first argument (got MyMemoryHandler instance instead)
 Error in sys.exitfunc:
 Traceback (most recent call last):
   File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
     func(*targs, **kargs)
   File "/usr/lib/python2.7/logging/__init__.py", line 1640, in shutdown
     h.close()
   File "/usr/lib/python2.7/logging/handlers.py", line 1110, in close
     logging.Handler.close(self)
 TypeError: unbound method close() must be called with Handler instance as first argument (got MyMemoryHandler instance instead)

Thanks.

zuozhengkun, 2016-04-18:
Hello,
  Any update?
   Have  you  encounter this  same  problem ?
My  current  python Version is   Python 2.7.3.

Or  it  need  to  be  upgrade  my python version ?
else  many  tools  depended on the python  will  be re-installed  later,  
python-pip python-dev ...
and then what  is  a  big-workload  to  handle  it !

Thank you very much.

zuozhengkun, 2016-04-18:
If  using  ctrl+C break, also  occur ,
---------->
KeyboardInterrupt
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/logging/__init__.py", line 1640, in shutdown
    h.close()
  File "/usr/lib/python2.7/logging/handlers.py", line 1110, in close
    logging.Handler.close(self)
TypeError: unbound method close() must be called with Handler instance as first argument (got MyMemoryHandler instance instead)
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/logging/__init__.py", line 1640, in shutdown
    h.close()
  File "/usr/lib/python2.7/logging/handlers.py", line 1110, in close
    logging.Handler.close(self)
TypeError: unbound method close() must be called with Handler instance as first argument (got MyMemoryHandler instance instead)

But  the  result  can  get :
./results/20160418_154747/rtapp:noeas:44_pct/1/trace.dat
./results/20160418_154747/rtapp:noeas:13_pct/1/trace.dat
./results/20160418_154747/rtapp:noeas:25_pct/1/trace.dat
./results/20160418_154747/rtapp:noeas:19_pct/1/trace.dat
....

derkling, 2016-04-28:
What is your setup?
Can you try to run the suite from within a VM... maybe by using the Vagrant file we provide at the root of the LISA tree.

To create a VM using vagrant you need to:
1.  install VirtualBox in your machine
2. install vagrant by downloading and installing the proper package from:
    https://www.vagrantup.com/downloads.html
3. create the vagrant VM using these commands:

```bash
$ cd $LISA_HOME
$ vagrant up
```

This command will download all the required packages and build a VirtualBox VM ready to be used.
To enter the VM, you need this command:

```bash
vagrant run
```

zuozhengkun, 2016-04-29:
Maybe  it  is   another  issue  about   this , 
after  test  it  would not  affect  the  run   result.

thanks.

mdigiorgio, 2016-05-24:
Is this still an issue to be considered **Open** or can we close it?

zuozhengkun, 2016-05-25:
OK,
 it  can be  closed  currently, and  if  an issue  yet , fire  another  case  again.
thanks.

closed 2016-05-25 by mdigiorgio