ARM-software/lisa #91: Failed to set CPUFreq ondemand governor's sampling_rate for Linux kernel 4.4

Leo-Yan, 2016-04-24:
Hi,

I saw Lisa set ondemand governor's sampling_rate in the folder: /sys/devices/system/cpu/cpufreq/ondemand/sampling_rate; but when I run Linux kernel 4.4 the folder has been changed as: /sys/devices/system/cpu/cpufreq/policy0/ondemand/sampling_rate and /sys/devices/system/cpu/cpufreq/policy1/ondemand/sampling_rate.

Thanks,
Leo Yan

mdigiorgio, 2016-04-25:
I think this is the same as #60, isn't it?

@jlelli was there a PR from you #62 addressing this?

jlelli, 2016-04-25:
I think this is a devlib problem. The fix I came up with in #62 should be general enough, assuming devlib doesn't use any hardcoded path (like I think it does).

mdigiorgio, 2016-04-25:
In fact, it is a devlib issue https://github.com/ARM-software/devlib/blob/master/devlib/module/cpufreq.py#L130

derkling, 2016-04-26:
@Leo-Yan (if not yet done) can you please open an issue on devlib to track that?
https://github.com/ARM-software/devlib/issues

Leo-Yan, 2016-04-26:
@derkling have filed this issue on devlib.

closed 2016-12-09 by bjackman