ARM-software/lisa #100: cgroups.ipynb : Fails to copy file to /root/schedtest

srinivas-kalaga, 2016-04-28:
I get the following error in cgroups.ipynb

"05:08:15  DEBUG   :          RTApp - Pulling logfiles to [/home/srinu/work/juno/HMP-linaro/lisa/results/20160428_170748]...
05:08:15  DEBUG   : adb -s 0123456789ABCDEF shell ls '/root/schedtest/*task0*.log'
05:08:15  DEBUG   : adb -s 0123456789ABCDEF pull '/root/schedtest/*task0*.log: Not a directory' '/home/srinu/work/juno/HMP-linaro/lisa/results/20160428_170748'"

This is because rtapp.conf(kind='profile', params=tasks, run_dir=TARGET_DIR) does not create the TARGET_DIR if it doesnot exist.

derkling, 2016-04-28:
I have a fix for that, going to push it now ;-)

derkling, 2016-04-28:
This should now be fixed in e3f02e72dc8937e3c8a8cb76cdaa615b7fcc0205
Could you please check and report?

srinivas-kalaga, 2016-04-28:
it works, thanks a lot.

closed 2016-04-28 by srinivas-kalaga