ARM-software/lisa #98: Error: cgroup error with Android

srinivas-kalaga, 2016-04-28:
I get 
"11:32:41  WARNING :         Target - Unable to initialize [cgroups] module
11:32:41  ERROR   :         Target - Fix your target kernel configuration or disable module from configuration"

Looks like the following code in target.py is responsible for this
 113     def config(self):
 114         try:
 115             return KernelConfig(self.execute('zcat /proc/config.gz'))
 116         except TargetError:
 117             for path in ['/boot/config', '/boot/config-$(uname -r)']:
 118                 try: 
 119                     return KernelConfig(self.execute('cat {}'.format(path)))
 120                 except TargetError:
 121                     pass
 122         return KernelConfig('')

derkling, 2016-04-28:
We have already merged a patch in devlib for that issue, but still not advanced the LISA version to track the devlib master. Since people seems to need it, I'm goint to update the submodule now.

derkling, 2016-04-28:
This should now be fixed by PR #99 
To test it you need to:
```bash
$ cd $LISA_HOME
$ git pull origin
$ source init_env
[LISAShell lisa] \> lisa-update
[LISAShell lisa] \> lisa-ipython stop
[LISAShell lisa] \> lisa-ipython start
```

srinivas-kalaga, 2016-04-28:
Thanks, it works now.

closed 2016-04-28 by derkling