ARM-software/lisa #105: Error in " nosetests -v tests/stune/smoke_test_ramp.py"

srinivas-kalaga, 2016-05-05:
schedtune tests fail as below

The boosted utilization signal is appropriately boosted ... ERROR

======================================================================
ERROR: The boosted utilization signal is appropriately boosted
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/srinu/work/juno/HMP-linaro/lisa/tests/stune/smoke_test_ramp.py", line 59, in test_boosted_utilization_signal
    boost_task_rtapp = sbt_dfr[sbt_dfr.comm == rta_task_name]
  File "/home/srinu/.local/lib/python2.7/site-packages/pandas/core/generic.py", line 2669, in __getattr__
    return object.__getattribute__(self, name)
AttributeError: 'DataFrame' object has no attribute 'comm'

----------------------------------------------------------------------
Ran 1 test in 52.432s

derkling, 2016-05-06:
@JaviMerino do you think the recent updates on Trappy could have affected the code for the SchedTune ramp test?

JaviMerino, 2016-05-09:
I've run the test and haven't been able to reproduce it.

srinivas-kalaga, 2016-05-09:
Thanks, will check my configuration again.

JaviMerino, 2016-05-10:
From looking at the traceback it looks like your trace doesn't contain any event for `sched_boost_task`.  Could it be that your kernel doesn't generate anything for that event?  Do you have patch "DEBUG: schedtune: add tracepoint for task boost signal" in your kernel? It should add a `sched_boost_task` tracepoint in `boosted_task_util()`

srinivas-kalaga, 2016-05-11:
Hi Javi,
Can you confirm if "DEBUG: schedtune: add tracepoint for task boost signal"
patch is part of EASv2 release as I dont see it on either Juno tree or our
internal kernel tree?
Thanks,
Srinivas

derkling, 2016-05-11:
Hi Srinivas,
all the debug tracepoints are not part of official RFC release on ML.

However, you can find all these patches are available in the repository and branch specified in the "Kernel Features" section of the README:
   https://github.com/ARM-software/lisa#kernel-features

Here they are:
   http://www.linux-arm.org/git?p=linux-pb.git;a=shortlog;h=refs/heads/lisa/debug

srinivas-kalaga, 2016-05-12:
Hi,
I get this error now, please check if its a know issue.

10:06:56  INFO    :       Executor -       /home/srinu/work/juno/HMP-linaro/lisa/results/20160512_100453
The boosted utilization signal is appropriately boosted ... ERROR

======================================================================
ERROR: The boosted utilization signal is appropriately boosted
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/srinu/work/juno/HMP-linaro/lisa/tests/stune/smoke_test_ramp.py", line 65, in test_boosted_utilization_signal
    task_start = boost_task_rtapp.index[0]
  File "/home/srinu/.local/lib/python2.7/site-packages/pandas/indexes/base.py", line 1256, in __getitem__
    return getitem(key)
IndexError: index 0 is out of bounds for axis 0 with size 0

thanks

derkling, 2016-05-12:
Can you please report here some example of "sched_boost_task" events you get in the textual trace.
These do not contain any sensible information regarding your platform but it can help a lot from our side to understand what's going on.

Something like that should work:
```shell
trace-cmd report trace.dat | grep sched_boost_task | head -n10
```

Which should return something like:
```text
       trace-cmd-1758  [005]  2764.424364: sched_boost_task:     comm=sudo pid=1757 util=897 margin=0
       trace-cmd-1758  [005]  2764.424375: sched_boost_task:     comm=sudo pid=1757 util=897 margin=0
       trace-cmd-1758  [005]  2764.424378: sched_boost_task:     comm=sudo pid=1757 util=897 margin=0
            sudo-1757  [002]  2764.426430: sched_boost_task:     comm=sh pid=1577 util=93 margin=0
            sudo-1757  [002]  2764.426433: sched_boost_task:     comm=sh pid=1577 util=93 margin=0
          <idle>-0     [005]  2764.427315: sched_boost_task:     comm=rcu_preempt pid=7 util=13 margin=0
          <idle>-0     [005]  2764.427320: sched_boost_task:     comm=rcu_preempt pid=7 util=13 margin=0
          <idle>-0     [005]  2764.427323: sched_boost_task:     comm=rcu_preempt pid=7 util=13 margin=0
              sh-1577  [000]  2764.427382: sched_boost_task:     comm=kworker/u12:2 pid=1586 util=0 margin=0
              sh-1577  [000]  2764.427387: sched_boost_task:     comm=kworker/u12:2 pid=1586 util=0 margin=0
```

srinivas-kalaga, 2016-05-16:
On Juno board
              sh-1673  [003]  3313.046759: sched_boost_task:     comm=sh pid=2213 util=1024 margin=0
              sh-1673  [003]  3313.046761: sched_boost_task:     comm=sh pid=2213 util=1024 margin=0
              sh-1673  [003]  3313.046763: sched_boost_task:     comm=sh pid=2213 util=1024 margin=0
              sh-1673  [003]  3313.046765: sched_boost_task:     comm=sh pid=2213 util=1024 margin=0
              sh-1673  [003]  3313.046766: sched_boost_task:     comm=sh pid=2213 util=1024 margin=0
              sh-1673  [003]  3313.046768: sched_boost_task:     comm=sh pid=2213 util=1024 margin=0
              sh-1673  [003]  3313.046771: sched_boost_task:     comm=sh pid=2213 util=1024 margin=0
              sh-1673  [003]  3313.046772: sched_boost_task:     comm=sh pid=2213 util=1024 margin=0
              sh-1673  [003]  3313.046774: sched_boost_task:     comm=sh pid=2213 util=1024 margin=0
              sh-1673  [003]  3313.046775: sched_boost_task:     comm=sh pid=2213 util=1024 margin=0

bjackman, 2016-12-09:
Looks like this is probably stale, can I close it?

bjackman, 2016-12-09:
(Clicked the wrong button, I cannot mouse)

JaviMerino, 2016-12-15:
I would say close it

closed 2016-12-15 by bjackman