ARM-software/lisa #106: aarch64 build for rt-app doesn't work

deggeman, 2016-05-09:
static build of json-c (json-c-0.11-20130402) with rt-app (https://github.com/scheduler-tools/rt-app.git scheduler-tools/exp/eas_v5 commit 1ebb50d234cd "Add configured run (c_run) to log metrics") doesn't work. I'm assuming here that these are the versions used for the arm (32bit) build in LISA under tools/arm64/.
tools/arm64 $ file rt-app
rt-app: ELF 32-bit LSB  executable, ARM, EABI5 version 1 (SYSV), statically linked

json-c_src: $ git reset --hard; git clean -fdx
json-c_src: $ ./autogen.sh
json-c_src: $ ./configure --host=aarch64-linux-gnu --disable-shared --enable-static
...
checking host system type... Invalid configuration `aarch64-linux-gnu': machine `aarch64' not recognized

deggeman, 2016-05-09:
It seems that commit fe1ac0201384 "Add automake/autoconf generated files for the 0.11 release." is the culprit.
97ef11033a39 - (tag: json-c-0.11-20130402, origin/json-c-0.11, _json-c-0.11-20130402, 97ef110) Drop the libtool versions back down a bit to avoid changing the major number of the library.  Programs compile
d1645e378d28 - Regen the doc pages using the right version for the 0.11 release.
4f4c3c17d474 - Bump the version numbers for the 0.11 release.
09c2e72b18e8 - Add the 0.11 release doxygen-generated docs to source control.
fe1ac0201384 - Add automake/autoconf generated files for the 0.11 release.
7ca1e523f05a - (HEAD, _7ca1e523f05a) Merge branch 'master' of https://github.com/json-c/json-c 

Using commit 7ca1e523f05a "Merge branch 'master' of https://github.com/json-c/json-c" works for an aarch64 build of json-c.

json-c_src: $ git reset --hard; git clean -fdx
json-c_src: $ git checkout 7ca1e523f05a
json-c_src: $ ./autogen.sh
json-c_src: $ ./configure --host=aarch64-linux-gnu --disable-shared --enable-static
json-c_src: $ make

rt-app_src $ git show --oneline
1ebb50d234cd Add configured run (c_run) to log metrics
...
rt-app_src $ git reset --hard; git clean -fdx
rt-app_src $ ./autogen.sh
rt-app_src $ ./configure --host=aarch64-linux-gnu LDFLAGS=" --static -L/opt/git/tools/json-c/.libs/" CFLAGS="-I /opt/git/tools/" --with-deadline
make

deggeman, 2016-05-09:
toolchain used: gcc-linaro-4.9-2014.11-x86_64_aarch64-linux-gnu

bjackman, 2016-12-09:
[image]

closed 2016-12-09 by bjackman