ARM-software/lisa #119: Add support to burn kernel/dtb with fastboot

Leo-Yan, 2016-05-31:
Hi there,

Could we provide the function to use fastboot to burn kernel and dtb file when reboot the board?

Thanks,
Leo Yan

derkling, 2016-05-31:
That would be a great feature which is already supported by devlib via its module::android::FastbootFlashModule
A better integration with LISA is however something which should be part of the TestEnv class.

For the time being we support only reboot via image copy, either via TFTP and/or by writing in the MMC card mounted by the host when a TC2/Juno board is rebooted.

valschneider, 2018-11-06:
With the LISA rewrite we've streamlined things and decided against kernel manipulation, so we also removed image copy.

closed 2018-11-06 by valschneider