ARM-software/lisa #124: Include a class diagram to help newcomers find their way around LISA

raw-bin, 2016-06-02:
This suggestion came about in discussions around reducing the learning curve with LISA. It would be neat to have some autogenerated class diagrams to show the overall design hierarchy. Should help especially if folk want to extend LISA.

Perhaps this can then find it's way on the wiki in an automated fashion ?

derkling, 2016-07-20:
Can this be something useful, or do we need to provide more details?
[image]

Maybe we can also add some links to the presentation and video I did at the Embedded Linux Conference.

derkling, 2016-07-20:
I've updated the main Wiki page... have a look if it could be enough (for the time being) to close this issue.

mdigiorgio, 2016-07-21:
Wiki looks nicer now :smile:

derkling, 2016-09-16:
No feedbacks collected so far.... thus let's assume this issue be solved for the time being.

closed 2016-09-16 by derkling