ARM-software/lisa #145: Vagrant: install wrapt

JaviMerino, 2016-07-12:
latest devlib depend on wrapt.  The vagrant file should `pip install wrapt`.  I'd rather do this once #113 is merged rather than adding even more stuff there.

bjackman, 2016-07-18:
Might it be easier to just use each lib's `setup.py`?

mdigiorgio, 2016-07-19:
Yes, probably we could just pip install `bart`, `trappy` and `devlib`... @derkling what do you think?

JaviMerino, 2016-07-19:
If we are going to pip install everything, then we should get rid of submodules.

derkling, 2016-07-19:
Tough  decision... one more time.
If it simplify deployment on VMs, perhaps it's better to pip install everything... however, we must be aware that the LISA shell by default uses "DEVMOVE"... thus it expects local installed submodules...

mdigiorgio, 2016-07-19:
OK, then let's just install dependencies when needed. I'll send a PR for solving this issue.

derkling, 2016-07-19:
> OK, then let's just install dependencies when needed. I'll send a PR
> for solving this issue.

Just to be clear, I'm open also to the other solution... only worried
for the changes required. We spent a lot of time to integrate
submodules... we just need to have a tought about what can be the
possible issues.

> 
> ---
> https://github.com/ARM-software/lisa/issues/145#issuecomment-233626911

mdigiorgio, 2016-07-19:
Adding things is simple. IMHO it is probably better to solve this by using tags and provide users with a stable release (that has no dependency or other issues). This way, dependency issues will hopefully only appear in development versions.

PR #152 (2016-07-21): Vagrant: install wrapt
The latest devlib depend on wrapt.  Install it.  This fixes #145.
commit ae6cf513 (2016-07-21): Vagrant: install wrapt

The latest devlib depend on wrapt.  Install it.  This fixes #145.
closed 2016-07-21 by mdigiorgio