ARM-software/lisa #207: about VM installation

cloverpop, 2016-10-08:
Hi,

I cloned lisa git repository as:
git clone https://github.com/ARM-software/lisa.git

Then I run "vagrant up" to finish the virtualbox instance downloading.
Once the VM instance is executed, it prompts to input login account and password.
Can we know what's the default login account and password with root permission？

Thanks.

bjackman, 2016-10-08:
Hi, 

You shouldn't need any username or password, it should automatically be set up with SSH key authentication, so it sounds like there's something wrong with your Vagrant installation. Maybe there's something in the log output when you create the box? It might be worth trying some of the things in this thread: https://github.com/puphpet/puphpet/issues/1253 (e.g. try bringing up a clean Vagrant box and seeing if the issue appears on that too, then you may be able to eliminate the Lisa Vagrantfile as the source of the problem).

cloverpop, 2016-10-09:
Thanks for your reply.

I tested in Windows VirtualBox Manager. Below settings now can work for me
acc = vagrant
pwd = vagrant

closed 2016-10-09 by cloverpop