ARM-software/lisa #208: lisa connection port

cloverpop, 2016-10-09:
Hi,

After I installed LISA VM instance in Windows OS and connected to LISA VM.
"vagrant ssh" command shows below connection info:

> 
> **`ssh` executable not found in any directories in the %PATH% variable. Is an
> SSH client installed? Try installing Cygwin, MinGW or Git, all of which
> contain an SSH client. Or use your favorite SSH client with the following
> authentication information shown below:
> 
> Host: 127.0.0.1
> Port: 2222
> Username: vagrant
> Private key: c:/zyang/share/lisa_src/.vagrant/machines/default/virtualbox/private_key**_
> 
> 

But when I run "source init_env" and "lisa-ipython start" in the LISA shell, it shows as below:

> Starting IPython Notebooks...
> Starting IPython Notebook server ...
>     IP Address : http://127.0.0.1:8888
>     Folder       :  /home/vagrant/lisa/ipynb
>     Logfile      : /home/vagrant/lisa/ipynb/server.log
>     PYTHONPATH : 
>                  /home/vagrant/lisa/libs/bart
>                  /home/vagrant/lisa/libs/trappy
>                  /home/vagrant/lisa/libs/devlib
>                  /home/vagrant/lisa/libs/wlgen
>                  /home/vagrant/lisa/libs/utils
> 
> Notebook server task: [1] 17472

Results:
1) http://127.0.0.1:8888
Windows Web Explorer shows:

> This site can’t be reached
> 
> The webpage at http://127.0.0.1:8888/ might be temporarily down or it may have moved permanently to a new web address.
> ERR_SOCKET_NOT_CONNECTED

2) http://127.0.0.1:2222
It shows error info as below：

> SSH-2.0-OpenSSH_6.6.1p1 Ubuntu-2ubuntu2.8
> Protocol mismatch.

Questions:
1) Which port is correct?
2) If there are potential firewall setting, can we configure port when we launch LISA notebook server?

Many thanks.

bjackman, 2016-10-10:
Hi,

8888 is the correct port for HTTP. 

I suggest testing the HTTP connection from inside the VM (`wget  http://127.0.0.1:8888` in your case), if that works then you know it's something to do with either port forwarding between the host and VM or the host firewall.

You may actually be the first person to run this on a Windows host. If you discover that it's a port forwarding issue then yes it is probably something we can fix in the Vagrantfile, if there's a host firewall setting that needs to be changed then it probably can't be done by Vagrant but we can document it.

bjackman, 2016-10-21:
Hi @cloverpop did you get this working? If so could you let me know how you solved it and I'll close the issue.

cloverpop, 2016-11-02:
Please close it.

closed 2016-11-02 by cloverpop