ARM-software/lisa #224: LISA energy model notebook fails to execute in Juno R0 board

cloverpop, 2016-11-03:
KeyErrorTraceback (most recent call last)
<ipython-input-15-9b1ffaa2f234> in <module>()
      2 loop_cnt = 5
      3 
----> 4 power_perf_df = compute_power_perf(clusters, loop_cnt, sysbench)

<ipython-input-14-0543b4f623b4> in compute_power_perf(clusters, loop_cnt, benchmark, bkp_file)
     77                         # Run benchmark into the target cgroup
     78                         perf += benchmark.run(target_cg.name, cnt + 1)
---> 79                         nrg = te.emeter.report(te.res_dir).channels
     80                         energy += nrg[cl.emeter_ch]
     81                         sleep(10)

/data/lisa_clean/lisa/libs/utils/energy.py in report(self, out_dir, out_file)
    204         for channel in self._channels:
    205             label = self._channels[channel]
--> 206             nrg_total = nrg[label]['total']
    207             logging.debug('%14s - Energy [%16s]: %.6f', 'HWMon', label, nrg_total)
    208             clusters_nrg[channel] = nrg_total

KeyError: 'a57'

cloverpop, 2016-11-03:
This issue can be assigned to me, please.

bjackman, 2016-11-04:
"juno" and "juno2" are actually to do with kernel versions i.e. the hwmon setup. The way these notebooks are intended to be used is that each user modifies some sections of it so that it works on their board - it's more of a guideline/example than a generic automated tool. `my_conf` is an example of this - we expect that each user will have to change the hwmon bits in my_conf.

This stuff isn't very user friendly at all, we definitely have some work to do on documentation. There's also a related issue here: https://github.com/ARM-software/lisa/issues/11 

But ultimately this is an issue with documentation and clarity, not with the notebook being broken, so I'm going to close this issue.

closed 2016-11-04 by bjackman