ARM-software/lisa #252: Define consistent documentation style/format

msrasmussen, 2016-12-13:
Documentation of many functions/classes is a bit sparse, but before that can be properly addressed a decision has to made on the in-code docstring documentation style. Looking through a few random python files I came across at least three different styles of documenting function arguments. In some cases there is even multiple styles used within a single file.
Could we have a decision on the in-code documentation style so it can be cleaned up and people help writing useful documentation?

bjackman, 2016-12-13:
Seems to me that [Sphinx autodoc](http://www.sphinx-doc.org/en/1.4.9/tutorial.html#autodoc) is the way forward for this. It's what Trappy and Bart use. I'll spend a little time investigating.

bjackman, 2016-12-20:
> we'll need to list the modules we want to document and how to structure that documentation (probably just mirror the Python package structure)

Actually seems we can just use `sphinx-apidoc` to structure this for us, that generates documentation for entire packages at a time. We'll still be able to add our own structure to things if we like.