ARM-software/lisa #255: Add to support tri-clusters

Leo-Yan, 2016-12-14:
Hi,

Now LISA has many code is to directly use 'big' and 'little' for plots, e.g. FrequencyAnalysis::plotClusterFrequencies() in file libs/utils/analysis/frequency_analysis.py. So can we change to use normal cluster naming like 'cluster' + id to present cluster 0, cluster 1, cluster 2, etc. So we can easily extend to support clusters. Does this feasible? And I'm curious with this kind fixing it will broke other stuffs?

If this is feasible, I can start to do some modification on X20 in my spare time.

Thanks,
Leo Yan

bjackman, 2016-12-14:
Yep I think this is certainly feasible. I've been [working on a](https://github.com/ARM-software/lisa/pull/246) more generic and detailed representation of platform topologies and hopefully we can then use that throughout LISA where possible. We certainly want to get rid of explicit big.LITTLE references whererever we can.

Leo-Yan, 2016-12-15:
@bjackman thanks for sharing the patches WIP. I saw there have many stuffs are WIP and I will check and apply them as needed and avoid to do duplicate things.

You could decide which time point is best to close this issue.

bjackman, 2016-12-15:
No worries. The next version is actually going to change the interface quite substantially (but isn't ready for sharing yet) so you might be better off doing some temporary hacks in the meantime. I will let you know as soon as there's something useful on Github.

derkling, 2016-12-15:
Hi @Leo-Yan, unfortunately the non bL support is not yet complete in LISA... however, you can probably get something working on an X20 board by using a platform configuration file
`libs/utils/platforms/x20.json` which contains something like:

```json
 {
     "board" : {
         "cores" : [
            "little", "little", "little", "little",
            "medium", "medium", "medium", "medium",
             "big", "big"
         ],  
         "modules" : ["cpufreq"]
     },
 }  
```

*NOTE:* that the "bl" module is not loaded.

With such a file you should be able to connect to the target using:

```python
import logging
from conf import LisaLogging
LisaLogging.setup(level=logging.DEBUG)

te = TestEnv({
       "platform" : "android",
        "board"   : "x20",
        "ANDROID_HOME" : "/path/to/your/android-sdk-linux"
    })
target = te.target
```

Which should build a `te.topology` like this:
```
cluster [[0, 1, 2, 3], [4, 5, 6, 7], [8, 9]]
cpu [[0], [1], [2], [3], [4], [5], [6], [7], [8], [9]]
```

This should be enough to control the target and run experiments using rt-app and the first CPU of the last cluster (i.e. CPU8) for calibration values. 

Once you collected a trace you can try to parse it via the `Trace` module and verify if some of its analysis methods fails because of hardcoded bL dependency... maybe they are not so much... but in that case you should be able to fix them by replacing the bl dependencies with the information reported by the topology object described before.

Leo-Yan, 2016-12-15:
Hi @derkling, at my side I wrote x20.platform as blow:

`
{
        // X20 boards have tri-clusters
        "board" : { 
                "cores" : [ 
                        "A53", "A53", "A53", "A53",
                        "A53", "A53", "A53", "A53",
                        "A72", "A72"
                ],
                "big_core" : "A72",
                "modules"  : [ "bl", "cpufreq" ]
        },

        "nrg_model" : { 
                "little" : { 
                        "cpu" : { 
                                "nrg_max" :   50, 
                                "cap_max" :  267 
                        },
                        "cluster" : { 
                                "nrg_max" :    6   
                        }
                },
                "big": {
                        "cpu" : { 
                                "nrg_max" :  444,
                                "cap_max" : 1024
                        },
                        "cluster" : { 
                                "nrg_max" :   23  
                        }
                }
        }
}
`

So this can 'cheat' LISA as it have two clusters, one is 8 cores and another is 2 cores :) I also used analysis method for task plots and cluster frequency plots, they could work if I only analyze one little cluster with CA72 cluster. 

I will align with your method; thanks for sharing this :)