ARM-software/lisa #323: AssertionError on Test First CPU

ryu614, 2017-03-03:
Hello, 

I am verifying EAS functionality on a new device by executing 

LISAShell lisa] \>lisa-test tests/eas/acceptance.py:ForkMigration

and I am having an error below. 

I’ve ran the above test on Google Pixel device, but I got same error as well. 

Am I missing something here?

 
[image]

bjackman, 2017-03-05:
Nope you aren't missing anything - the LISA framework is likely behaving correctly here, i.e. the ForkMigration test is asserting that all newly-forked tasks should run on a big CPU in a big.LITTLE system. Howerver this test is part of an older suite - starting every new task on a big CPU is not necessarily the expected behaviour for EAS. The plan is to remove acceptance.py in the near future.

To verify that the framework is behaving correctly, you can take a look at the trace file that was printed in the output, either using the kernelshark tool or using the `plot_trace` function in a LISA notebook - see `ipynb/tutorial/05_TrappyUsage.ipynb` for an example usage of that API (you'll need to actually run the notebook locally to see the trace plot, GitHub doesn't display the Javascript widget it uses). If you see that the workload threads were forked onto one of the "smaller" CPUs then the framework is working as expected.

As for verifying EAS behaviour, let's talk about that via another channel :)

bjackman, 2017-03-14:
The ForkMigration test is asserting that all newly-forked tasks should run on a big CPU in a big.LITTLE system. Presumably your x86 system is not big.LITTLE-like, in which case this test should be expected to fail.

For verifying EAS funcionality I'd point you toward all the other tests under `tests/eas/`. I'm really interested to know more about your use-case and goals, are you looking at deploying EAS on an x86 platform?

Please be aware that LISA has seen scant testing/support on x86 systems so you are likely to see issues and need to get familiar with the LISA code and make changes (contributions are welcome!). [Here's](https://github.com/ARM-software/lisa/pull/256) one outstanding x86 issue that I'm aware of.

kesavanabhilash, 2017-03-14:
I have an atom based platform and was playing around with the eas patches to see if they would save me some power. This is an octa core SMP system and I thought the better task packing in eas should yield some power savings (even on my SMP). I am referring to the ARM hikey platform code (energy model) to make the requisite changes for my platform.
I was under the impression that LISA acceptance tests would also test eas out on SMP systems. Should I try simulating a bL system by capping the max frequency of some cores to a lower one than the others  ? Would that be a better way to run the acceptance test or should I just drop doing that ?
I see several tests such as capacity capping, load tracking, rfc etc in the tests/eas folder, would all of them run on an SMP system ?
Thanks for the pointers regarding existing x86 issues, will dig into the LISA code some more. Also, is there some other forum where I can get my doubts clarified or is this it ?
Thanks.

bjackman, 2017-03-14:
Cool! I'd be interested to hear the results of whether EAS's packing reduces energy on an Intel system - believe Intel engineers have expressed the view that it won't (because of 'race to idle', i.e it's better to complete all your activations ASAP and then go to a deep idle state) but I haven't seen any measurements. 

> I was under the impression that LISA acceptance tests would also test eas out on SMP systems. Should I try simulating a bL system by capping the max frequency of some cores to a lower one than the others ? Would that be a better way to run the acceptance test or should I just drop doing that ?

Yeah, I would just recommend dropping that - those tests are totally bL specific.

>  I see several tests such as capacity capping, load tracking, rfc etc in the tests/eas folder, would all of them run on an SMP system ?

Yeah, IIRC they should all work on an SMP system except for acceptance.py.

One of the things worth mentioning is that EAS largely works on the assumption that it's working with a `sched`/`sched_util` cpufreq governor. I don't know how well this assumption translates to Intel platforms given the `intel_pstate` system. A similar issue is that LISA assumes it can use the `userspace` governor to set frequencies, I'm not sure whether that really works with Intel systems.

>  Thanks for the pointers regarding existing x86 issues, will dig into the LISA code some more. Also, is there some other forum where I can get my doubts clarified or is this it ?

I would say the main forum for discussing EAS is the [eas-dev mailing list](https://lists.linaro.org/mailman/listinfo/eas-dev) that Linaro hosts.