ARM-software/lisa #327: stune error

kesavanabhilash, 2017-03-10:
Hi,

I am trying to run eas on an x86 platform and am using LISA for verification of the changes made. When I run the schedtune smoke_test_ramp test I get the following error:

 ======================================================================
ERROR: test suite for <class 'tests.stune.smoke_test_ramp.STune'>
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/nose/suite.py", line 209, in run
    self.setUp()
  File "/usr/lib/python2.7/dist-packages/nose/suite.py", line 292, in setUp
    self.setupContext(ancestor)
  File "/usr/lib/python2.7/dist-packages/nose/suite.py", line 315, in setupContext
    try_run(context, names)
  File "/usr/lib/python2.7/dist-packages/nose/util.py", line 471, in try_run
    return func()
  File "/home/test/lisa/tests/stune/smoke_test_ramp.py", line 62, in setUpClass
    super(STune, cls).runExperiments(*args, **kwargs)
  File "/home/test/lisa/libs/utils/test.py", line 104, in runExperiments
    cls.executor.run()
  File "/home/test/lisa/libs/utils/executor.py", line 260, in run
    if not self._target_configure(tc):
  File "/home/test/lisa/libs/utils/executor.py", line 458, in _target_configure
    return self._setup_cgroups(tc)
  File "/home/test/lisa/libs/utils/executor.py", line 401, in _setup_cgroups
    self._setup_controller(tc, controller)
  File "/home/test/lisa/libs/utils/executor.py", line 420, in _setup_controller
    self._setup_group(tc, group)
  File "/home/test/lisa/libs/utils/executor.py", line 427, in _setup_group
    group.set(**tc['cgroups']['conf'][kind][name])
  File "/home/test/lisa/libs/devlib/devlib/module/cgroups.py", line 294, in set
    .format(self.controller.kind, attr_name))
ValueError: Controller [schedtune] does not provide attribute [schedtune.boost]

----------------------------------------------------------------------

On investigating this, the error seems to occur because the following command fails:
adb -s <device id> shell echo '[ -d /data/local/tmp/devlib-target/cgroups/devlib_cgh2/stune ] || mkdir -p /data/local/tmp/devlib-target/cgroups/devlib_cgh2/stune' | su

When I try to manually create the stune directory in the devlib_cgh2 directory (where the schedtune cgroup is mounted), I get a "no space left on device" message.  If a try to create a file in the devlib_cgh2 directory gives a "permission denied" error. Any pointers would be appreciated.

derkling, 2017-03-11:
SchedTune supports only a limited number of boostgroups to be created concurrently, which is 5 by default. If you happen to have already 5 boostgroups (i.e. SchedTune cgroups/forlder) then this can be a reason why you get a "no space left on device" for each successive mkdir operation.

Can you please send the output of this command:

```shell
cat /proc/cgroups
```

kesavanabhilash, 2017-03-12:
Thanks for the reply Patrick. Please find below the output of cat /proc/cgroups:

#subsys_name    hierarchy       num_cgroups     enabled
cpu     3       2       1
cpuacct 1       35      1
schedtune       2       4       1
freezer 4       1       1
debug   4       1       1

kesavanabhilash, 2017-03-12:
I just check my kernel sched/tune.c file and see that BOOSTGROUPS_COUNT is defined as 4. Does this mean that I have already hit the limit since proc/cgroups also shows 4 ?

kesavanabhilash, 2017-03-13:
Changing the BOOSTGROUPS_COUNT  to 16 fixes the error for me.

derkling, 2017-03-14:
Can I ask you why you are trying EAS and SchedTune on an x86 platform?
Can you share some details of your use-case... seems interesting ;-)

kesavanabhilash, 2017-03-14:
Patrick, I just replied to Brendan on #323.

closed 2017-03-16 by kesavanabhilash