ARM-software/lisa #371: EAS Testing for OnePlus3T - error running generic.py

RenderBroken, 2017-04-19:
I am working on implementing EAS for the OP3T which uses the same SD821 SoC. So with that I have been able to setup the testing environment but I am running into an issue when running:

lisa-test tests/eas/generic.py

It will get through the testing and give me some information.  But this error, I know nothing about. I tried to research it but I am apparently lacking in python and panda  data frames. if you have time, i would love any assistance. Here it is:

Traceback (most recent call last):
  File "/home/renderbroken/lisa/libs/utils/test.py", line 219, in experiment_test
    wrapped_test(experiment, tasks, *args, **kwargs)
  File "/home/renderbroken/lisa/tests/eas/generic.py", line 513, in test_task_placement
    self._test_task_placement(experiment, tasks)
  File "/home/renderbroken/lisa/tests/eas/generic.py", line 291, in _test_task_placement
    exp_power = self.get_expected_power_df(experiment)
  File "/home/renderbroken/lisa/tests/eas/generic.py", line 256, in get_expected_power_df
    task_utils_df.apply(exp_power, axis=1))
  File "/usr/local/lib/python2.7/dist-packages/pandas/core/frame.py", line 4152, in apply
    return self._apply_standard(f, axis, reduce=reduce)
  File "/usr/local/lib/python2.7/dist-packages/pandas/core/frame.py", line 4248, in _apply_standard
    results[i] = func(v)
  File "/home/renderbroken/lisa/tests/eas/generic.py", line 251, in exp_power
    expected_utils = nrg_model.get_optimal_placements(task_utils)
  File "/home/renderbroken/lisa/libs/utils/energy_model.py", line 678, in get_optimal_placements
    power = self.estimate_from_cpu_util(util, freqs=freqs)
  File "/home/renderbroken/lisa/libs/utils/energy_model.py", line 613, in estimate_from_cpu_util
    freqs, idle_states, combine=True)
  File "/home/renderbroken/lisa/libs/utils/energy_model.py", line 553, in _estimate_from_active_time
    _idle_power = max(node.idle_states[idle_states[c]] for c in cpus)
  File "/home/renderbroken/lisa/libs/utils/energy_model.py", line 553, in <genexpr>
    _idle_power = max(node.idle_states[idle_states[c]] for c in cpus)
KeyError: ('C2', u'occurred at index 0.448395')

Kind Regards,
Zachariah

bjackman, 2017-04-19:
Hi Zachariah,

Cool to hear that you're trying LISA out on a new device! I guess you used the EnergyModel that LISA provides for Pixel?

I suspect this error is due to a difference between the Pixel and OnePlus kernels in how CPU IDs are reported in ftrace. I'll try to find time to stare at the code and figure it out, then get back to you with more questions :)

RenderBroken, 2017-04-20:
Actually, please disregard. I found it was a typo error on my side. I forgot to rename a certain line from pixel to op3:

-pixel_energy = EnergyModel(
 +op3_energy = EnergyModel(

Here is a commit that shows more info:

[https://github.com/RenderBroken/lisa_toolkit/commit/564c88691d1276b4f1d8ee23e8f16e84f169c169](url)

Thanks again though for the amazing fast response! I am hoping to contribute what little I can to this project. I am weak at python and really new to all of this but I am sure I could possibly help with documentation. Getting Lisa cloned to working like it is today was a journey. Maybe I can make that easier for others to be able to give it a shot and hopefully contribute as well. 

Kind Regards,
Zachariah

closed 2017-04-20 by RenderBroken