ARM-software/lisa #385: android: Add monkey-test workload

joelagnel, 2017-05-07:
monkey tests are very useful doing lot of activity in short period of time on the device and collecting traces to analyze frame drops and other behavior. It would be nice to have a workload in workloads/ do this. I have something I'll post soon, this issue tracks it.

bjackman, 2017-05-08:
It's worth noting that Workload Automation might be able to solve this use case already (some of the stuff we've been doing in LISA has been just because WA wasn't working with the latest Android, which may now be fixed).

joelagnel, 2017-05-08:
Ok. maybe we're talking of different things? - this is basically the monkey apk launching an app and doing lots of activities (user interaction) to it, like 1000s of random touch, swipe events etc. The workload would essentially just call monkey.

joelagnel, 2017-05-09:
@bjackman assign to me? (also let me know if/how I can assign to self)

douglas-raillard-arm, 2020-06-05:
Closing as workloads are now in WA

closed 2020-06-05 by douglas-raillard-arm