ARM-software/lisa #386: android: Add camera workload

joelagnel, 2017-05-08:
This feature is for tracking addition of camera workload to android.

joelagnel, 2017-05-08:
How do I assign an issue to myself?

joelagnel, 2017-05-09:
A simple test could just do the following:
        am force-stop com.google.android.GoogleCamera;
        sleep 1;
        am start -W -S -n com.google.android.GoogleCamera/com.android.camera.CameraLauncher;
        sleep 6;
        am start -a android.intent.action.MAIN -c android.intent.category.HOME;
        sleep 4;
        am force-stop com.google.android.GoogleCamera;
        sleep 1;

Later on we can add ability to take a camera shot and such things.

valschneider, 2018-02-23:
https://github.com/ARM-software/lisa/pull/447

closed 2018-02-23 by valschneider