ARM-software/lisa #387: Add support/abstraction for running command-line scripts with LISA

joelagnel, 2017-05-09:
@derkling @bjackman , just created this to track this feature. It is useful to have an abstraction to run LISA scripts from the command line to parse common command line arguments and provide a more well defined way to writing these scripts.

joelagnel, 2017-05-09:
I see this already there, maybe we want to improve/extend this?
https://github.com/ARM-software/lisa/blob/master/libs/utils/android/benchmark.py

joelagnel, 2017-05-09:
Also, may be we can have a generic benchmark collector which you can pass a optional conf if needed?
So I should be able to say:
benchmark --workload=youtube --collect=energy 

I see benchmarks are their own python files, which call into workload.py (https://github.com/ARM-software/lisa/tree/master/tests/benchmarks), do we really need a separate python wrapper for each workload? CC: @ionela-voinescu @jlelli

valschneider, 2018-11-06:
See https://github.com/ARM-software/lisa/pull/636

closed 2018-11-06 by valschneider