ARM-software/lisa #416: libs/utils/analysis should make separation between plotting and analysis more clear

joelagnel, 2017-06-09:
Currently LISA's libs/utils/analysis/ is more geared towards plotting. I think this limits its reuse to just collect some data without having to plot. For example sometimes it may be only desired to get a single number as an analysis result.

Even analysis results in tabular form can actually provide as a DataFrame result instead of the analysis module trying to plot it. This also makes the analysis part of the code hard to read as it interferes with the plotting logic. I feel the plotting logic should be separate and independent of the analysis logic.
Then the user can do whatever they want with those results. The granularity should be at that level and the plotters should be in their own modules. Probably libs/utils/analysis/plot/ or something along those lines. Even if a new module isn't created, I feel we should break down functions into smaller units and do the plotting separately in separate function. This will make LISA more useful as well in the long run.

I also feel the same about the Trace class. It has too many things about CPU frequency and other assumptions in it. I feel it shouldn't be mandatory to pass a platform parameter there if someone just wants to create a trace object.

joelagnel, 2017-06-09:
So I saw some modules are better written in that the separation is better, seems it uses _dfg prefix, I guess that means data frame getters? That sounds like more what I talking about.
I think the separation is not that great though in cpus_analysis.py so that's what mostly triggered this bug request.

joelagnel, 2017-06-10:
Hi @derkling so I spent 2 hours to understand your analysis scripts and I really like what you're doing with the data frame getters. I think this approach would work very well for the scripts. Kudos for the clean design!
I am thinking of adding an 'AnalysisTool' class under libs/utils/analysis/tools/analysis_tool.py to parse common command line arguments (just 1 for now, that is --trace-file) and then create some tools to subclass this and the sub classes would parse tool specific arguments. Did you have any thoughts about that? I would then want to create some tools that people can pass trace files to and get analyses results.
I am open to design ideas and I am sure a lot of people will love LISA for its ability to do this. CC @bjackman

bjackman, 2017-06-12:
I think I agree with everything in here. Would love to get rid of the need for a platform parameter in the `Trace` class or at least to make it optional.

As for the separation of analysis and plotting I also agree - the `_dfg` methods are a nice big step in that direction as you said. The main thing that's missing there is documentation I think. Did you know that all the `_dfg_foo` methods get translated into `trace.data_frame.foo` methods? It's really useful but currently a little difficult to explain & document.

As for command line tools, I think that's a nice idea. One thing worth considering there is that we might want to the input format to those tools to be not just a trace file, but actually a directory with a trace file plus some auxiliary stuff (the most obvious auxiliary thing is platform.json but it's easy to think of other things that might want to go in there). So we might need to think about that in the interface design. On the other hand, I am reluctant to create a rich/complex command-line interface given that anything rich & complex is probably just best done as a notebook. Also before I got involved in the project I think there were some things like this so would be worth looking at them (@derkling will know more about this)...

joelagnel, 2017-06-12:
Hi @bjackman, Agreed with making CLI simple and leaving rich things for notebooks. Yes I knew about that dfg data frame feature, it's  v. Intuitive ;-) About providing trace directory, LISA is now integrated into Android tree and lot of people I hope will start using it. Many of them don't know LISA and would just want to collect something with systrace and parse it and grab some numbers. Considering this I think it's beneficial to provide a trace file option along with an option for directory if needed. I would love to hear what @derkling already has as you mentioned.

bjackman, 2017-06-12:
> I think it's beneficial to provide a trace file option along with an option for directory if needed

Yeah, makes sense - it could operate on traces by default and directories where they're provided.

valschneider, 2018-12-28:
We explicitely use df_* & plot_* in LISA/next, so that should be clear enough.

closed 2018-12-28 by valschneider