ARM-software/lisa #419: Apply this tool to Snapdragon 835.

AndyHeo, 2017-06-15:
Hi,

I want to apply this tool to Snapdragon 835(two clusters, 4 big cores, 4 little cores) for testing android device(EAS).
But this tool provides service to few devices and can't proceed on my device.  
If I want to test my device, which part to modify on this source?
(ex. energy.py, libs/utils/platforms/)

Environment
Kernel Ver. : 4.4
Android : O - OS

Thanks, 
Andy Heo

valschneider, 2017-06-15:
Hi,

When you setup your TestEnv instance with the configuration dictionary, LISA will use the 'board' value to find an energy model in **libs/utils/platforms**, e.g.
"board" : "hikey960" => it will look for **libs/utils/platforms/hikey960.json**. If no file is found, it will attempt to read the energy model directly from the target - you must have the **cpuidle** and **cpufreq** devlib modules enabled.

"platform" : <either linux or android>
"board" : <whatever name you want to give to your board>
"modules" : [ "cpufreq", "cpuidle" ]

I think you still need to have a board.json file for some analysis/plotting. I have a patch that outputs that json file using the energy model read on the device, I could send it to you if you want. In the longer run I would like that to be somewhat automated so new boards require no/minimal modifications.

AndyHeo, 2017-06-16:
I tried to fix it myself, but it was blocked by emeter function.
I would be very grateful if you share the patch.

Thank you for your response.

valschneider, 2017-06-16:
The two commits at the top of the branch should do the job: https://github.com/valschneider/lisa/tree/better_nrg
It's mostly a kludge, but should work for now.

AndyHeo, 2017-06-19:
After applying the patch you suggested, the error message occurs.
2017-06-19 14:28:57,195 INFO    : TestEnv      : Attempting to read energy model from target
2017-06-19 14:29:02,786 ERROR   : TestEnv      : Couldn't read target energy model: Active states powers are expected to be monotonically increasing. Values: [56, 76, 91, 105, 118, 135, 150, 162, 181, 196, 214, 229, 248, 280, 316, 354, 392, 439, 495, 565, 622, 691, 792, 889, 1024, 1141, 1138]

We think that the energy model was not created due to the above error.
Could you guide me how to fix this issue?

For you information, the configuration and test example are like below.
# Setup a target configuration
my_conf = {
    
    # Target platform and board
    "platform"    : 'android',
    "device"    : '2fe45ad0',
    
    # Target board IP/MAC address
    "host"        : '10.1.211.18',
    
    # Login credentials
    "username"    : 'root',
    "password"    : '',
    
    # Tools required by the experiments
    "tools"   : ['trace-cmd'],
    "modules" : ['bl', 'cpufreq', 'cpuidle', 'hotplug', 'cgroups'],
    
    # FTrace events to collect for all the tests configuration which have
    # the "ftrace" flag enabled
    "ftrace"  : {
         "events" : [
            "cpu_frequency",
            "cpu_idle",
            "sched_switch"
         ],
         "buffsize" : 10 * 1024,
    },
}

example
EnergyModel_ClusterEnergy.ipynb

valschneider, 2017-06-19:
That's a fairly simple issue: As the error message says, active states powers must be monotonically increasing, so the last two values should be swapped. That would be found in the device's energy model in the device tree.

If you haven't done it already I'd recommend reading [the EAS overview](https://developer.arm.com/-/media/developer/developers/open-source/energy-aware-scheduling/eas_overview_and_integration_guide_r1p2.pdf?la=en) - the last few pages are on the device tree entries.

credp, 2019-02-21:
Nobody looked here since Jun 2017, so I think it's safe to say this is resolved.

closed 2019-02-21 by credp