ARM-software/lisa #422: Make tests/benchmarks/ more useable

joelagnel, 2017-06-17:
Currently tests/benchmarks does stuff like automatically testing governor's etc. I don't see a point in enforcing this on the user (for example if a user wants to just run a workload and collect some traces). Also several conf related things are hard coded in the scripts.
Instead the approach should be:
- provide command line options for test config option
- automate as much of the config detection as possible (example read serial number automatically if only one device connected) so config is minimal 
- provide command line options for collect workload parameter (systrace, energy etc)

Could we just take out all the governor stuff from​benchmarks?

Also one more idea - any test specfic tuning can be done by the user subclassing the benchmark specific class and override hooks to run pre and post benchmark stuff and also wrap the base benchmarks' run method.
So the class hierarchy if someone wants to hack in governor stuff should be:
UibenchGovernor -> UiBenchmark -> LisaBenchmark

And the governor specific stuff can just live in another directory without disturbing the main use case. Thought?

joelagnel, 2017-06-18:
Actually I just found that '--collect' is already an argument, so sorry about that.

derkling, 2017-06-19:
Regarding governors removal I would say not, at least for the tests we have so fare. Indeed this was the main purpose we added these tests. 

What should be possible instead is to re-factor existing tests in such a way that the testing with different governors is a subtest of a base on which er just run the workload one time.

For example, we should update `tests/benchmarks/android_jankbench.py` to provide a base class which just run one time a workload. But in the same PR we should add another test, e.g.
`tests/benchmarks/android_jankbench_governors.py` which derives from `JankbenchTest` and basically provides the current implementation to run it with all the possible available governors.

valschneider, 2018-11-06:
I think it's safe to say wltests has replaced those benchmarks we had.

closed 2018-11-06 by valschneider