ARM-software/lisa #441: Add feature to find nr_running on a CPU

bjackman, 2017-07-10:
It can be useful to know the length of a CPU's runqueue at any given moment. Should add an analysis feature to do this.

bjackman, 2017-07-10:
Work in progress here, just sharing in case it can be useful to anyone: https://github.com/bjackman/lisa/tree/trace-nr-running

Broadly seems to work, needs validating and probably some tweaks for corner cases (duplicate indices, values at the beginning of the trace before we have data).

joelagnel, 2017-07-15:
Nice I love this! Another great analysis I think could be to check how
often sched_stat_runtime gets updated for a task so that we can identify
issues related to the lack of it's update.

bjackman, 2017-07-17:
Cool. BTW the end goal of this analysis is to figure out how much time there was when one more CPUs had multiple runnable tasks while at the same time another CPU was idle.

bjackman, 2017-07-17:
Oops, fat-fingered the close button

bjackman, 2017-07-20:
Bit more progress, the first step of this task is to figure out which which rq tasks were attached to: I've got a pretty solid (read: there is 1 unit test :P) implementation here: https://github.com/bjackman/lisa/commit/5e0ca5e09a6d5ab883a766c7bc87dac7f729ad99

Next step is to:

- Figure out when each task was runnable (shaky WIP here: https://github.com/bjackman/lisa/commit/0cf49d0a33dc53e6fb30d7f0a3b0b8c65a9320a1)
- Combine info about tasks being runnable with which rq they're attached to to give you nr_running (shaky WIP here https://github.com/bjackman/lisa/commit/d1806f1b3f6c38e767117335ce7ff208b7a17aef)

closed 2020-06-05 by douglas-raillard-arm