ARM-software/lisa #449: Bug: RTA property delay_s will case kernel segmentation fault

cloverpop, 2017-07-21:
rt-app workload with delay_s property triggers kernel fault immediately.

[28081.878055] delay_starter_1[4059]: unhandled level 0 translation fault (11) at 0x5f746c74
[28081.895414] pgd = ffffffc0b5e9b000
[28081.898809] [5f746c75616665] *pgd=0000000000000000, *pud=0000000000000000
[28081.905976] 
[28081.907552] CPU: 5 PID: 4059 Comm: delay_starter_1 Not tainted 4.4.77-g576a011-dirty #1
[28081.915561] Hardware name: HiKey960 (DT)
[28081.919485] task: ffffffc0787ce080 ti: ffffffc094944000 task.ti: ffffffc094944000
[28081.926973] PC is at 0x43c6c4
[28081.929941] LR is at 0x4028b4
[28081.932904] pc : [<000000000043c6c4>] lr : [<00000000004028b4>] pstate: 00000000
[28081.940298] sp : 0000007fb7ffd610
[28081.943608] x29: 0000007fb7ffd610 x28: 00000000004b16f0 
[28081.948925] x27: 0000007fb7ffe0a0 x26: 00000000004af000 
[28081.954239] x25: 0000000000402ca8 x24: 00000000004a9000 
[28081.959560] x23: 00000000004a8000 x22: 0000007ffffff65f 
[28081.964877] x21: 0000000000000000 x20: 0000000000000000 
[28081.970191] x19: 00000000004a84a0 x18: 0000000000000000 
[28081.975512] x17: 00000000004a9000 x16: 0000000000000000 
[28081.980828] x15: ffffffffffffffff x14: 0000000000000001 
[28081.986141] x13: 0000000000000010 x12: 0101010101010101 
[28081.991464] x11: 7f7f7f7f7f7f7f7f x10: fefefefefeff5c1f 
[28081.996780] x9 : ffffffffffffffff x8 : 705f7413d1617eb5 
[28082.002095] x7 : 0000808080808080 x6 : 0000007fa400190a 
[28082.007417] x5 : ffffffffffffffff x4 : 0000000000005d20 
[28082.012733] x3 : 705f746c75616665 x2 : 000000000000007f 
[28082.018048] x1 : 0000007fa40018d0 x0 : 705f746c75616665

bjackman, 2017-07-21:
This is a bug in the kernel... looks like it was on hikey960 so presumably it's the linaro hikey kernel, might be worth reporting it to Linaro with the full backtrace (showing the debug symbols)

cloverpop, 2017-07-23:
No, this kernel fault is triggered by a wrong rt-app json file, which is generated by LISA tool.
If we correct the LISA scripts to generate the correct json file that can meet the latest RT-APP requirement for "delay" property, this kernel fault is gone.

cloverpop, 2017-07-23:
Patch was merged

closed 2017-07-23 by cloverpop

bjackman, 2017-07-23:
> No, this kernel fault is triggered by a wrong rt-app json file

A bad rt-app json file should not be able to trigger a kernel fault - the bug in LISA (which is now fixed) is that it produced a bad rt-app JSON file, but there's still a (separate) bug in the kernel.

derkling, 2017-07-24:
The reported backtrace is not just a dump on the klog of a user-space crash?

The PC is at 0x43c6c4, which is within the user-space addressing space:
   https://www.kernel.org/doc/Documentation/arm64/memory.txt