ARM-software/lisa #516: Clarify/document EnergyMeter interface

cdleonard, 2017-12-04:
I wrote a EnergyMeter implementation for an in-house power measurement board and I don't know if I did it correctly. Since the board is not going to be available externally it wouldn't be useful to post it as a patch. Existing implementations don't apply: my meter has a microcontroller connected via a separate usb/serial and outputs human-readable text that I parse with regexps.

It would help to expand the documentation on the EnergyMeter class. I read https://github.com/ARM-software/lisa/wiki/Energy-Meters-Requirements but still had trouble with the following:

- It's not clear what report() should return.
  - I returned milijoules in the channel dict. Other implementations also seem to returning energy (not power), but maybe returning joules as floats is important?
  - I returned None as EnergyReport data_frame. What does this affect?
- sample() seems to be returning a dict of dict with "last"/"delta"/"total" since last call and values seem to be joules again. I did not implement this, what does it affect?
- Maybe I should have implemented a devlib.Instrument? I see no obvious benefit.

I'm not sure if this is the proper medium to ask these questions.

mdigiorgio, 2017-12-05:
Hi @cdleonard, thanks for using our tool. You are right, we should document the `EnergyMeter` class at some point.  Here some answers to your questions:

- `report()` should return an `EnergyReport` namedtuple with 3 elements:
    * `channels`, a dictionary that reports the energy read from each channel
    * `report_file`, path to the file containing all samples read from `reset()`
    * `data_frame`, a pandas `DataFrame` with all samples read from `reset()`

  The unit of measure is not enforced, it usually depends on the meter you are using. We usually convert the string read from the meter to a float (see the `ACME EnergyMeter`), but you can convert it to what you think is best for your post-processing tools.
  As long as you do not use the `data_frame` field, if you set it to `None` it will not affect anything.

- `sample()` is an optional method that can be used for example to implement a sort of polling strategy for continuously reading and displaying data. You can safely avoid to implement it if you are only interested in reading the total energy consumption from `reset()` to `report()`.

- Based on what you said about your setup I think implementing and `EnergyMeter` is enough.

Hope this helps. Should you have any other questions do not hesitate to ask.

bjackman, 2017-12-05:
FWIW, the reason for implementing `devlib.Instrument` would be that the upcoming [version 3 of Workload Automation](https://github.com/ARM-software/workload-automation) (currently in dev in the [in the 'next' branch](https://github.com/ARM-software/workload-automation/tree/next)), is based on devlib. If you implemented it in devlib then adding support to WA would be very easy. If you don't need WA support, implementing it directly in LISA is absolutely fine.

cdleonard, 2018-01-04:
My question has been answered so there is no reason to keep this open.

It would be nice for this to be documented better though, including exactly what keys the dicts should contain and what values should be in that DataFrame.

closed 2018-01-04 by cdleonard