ARM-software/lisa #529: nrg_model confusion

cdleonard, 2017-12-18:
The ["integrating a new board in LISA"](https://github.com/ARM-software/lisa/wiki/Integrating-a-new-board-in-LISA) page claims only a few fields are required but trying to run with such a bare config results in confusing problems (python exceptions).

In particular it seems the TestEnv.platform dict needs an "nrg_model" which seems to partially duplicate EAS energy-costs from DTS? In TestEnv._init_target there is apparently an attempt to read the energy model from target. Unfortunately TestEnv.nrg_model seems to be different from TestEnv.platform['nrg_model'] and the latter can only be loaded from json inside _init_platform.

There are various check for 'nrg_model' in TestEnv.platform in libs/utils/trace.py which makes me think this is optional. But looking at TestEnv._init_platform when init fails it just leaves a None value in the platform dict. I had some luck with the following patch:

```
diff --git libs/utils/env.py libs/utils/env.py
index 013f44f..5ea783b 100644
--- libs/utils/env.py
+++ libs/utils/env.py
@@ -797,6 +797,9 @@ class TestEnv(ShareState):
         else:
             self.platform['nrg_model'] = self._load_em(self.conf['board'])
 
+        if self.platform['nrg_model'] is None:
+            del self.platform['nrg_model']
+
         # Adding topology information
         self.platform['topology'] = self.topology.get_level("cluster")

```

This seems like an ugly hack.

Perhaps the EnergyModel object fetched from the target should be converted to the format that is expected inside platform['nrg_model']? But ideally when no nrg_model is available the whole framework should still run, yes?

derkling, 2017-12-21:
Hi Leonard, thanks for this bug report!

The energy model story within LISA have evolved along time in two different concepts.

Originally there was only `self.platform['nrg_model']`, which is loaded from JSON by TestEnv._load_em(self.conf['board']). As you correctly point out, that's just a "simplified" EM which reports few key values (e.g. max capacity of LITTLE CPUs) and which is used mainly to embellish some of the plots we can generate with the analysis module.

More recently, we added the support to read an EM from a device's DT, thus generating a complete Python representation of the in-kernel EM data. This representation however does not currently support serialization on disk and it's used mainly to support EAS related tests (i.e. `tests/eas/generic.py`).
When tests are executed we load the EM from the device and use it to validate scheduler decisions based on the optimal expected tasks allocation.
Unfortunately, we did not yet had bandwidth to work on the generation of a simplified JSON EM from this one loaded from the device. This would allow us to consolidate the two EM representation in just one.

Regarding your issue, you are right: TestEnv should work also if a simplified JSON EM is *not* available.
Hence the multiple checks we have int TestEnv, still it seems that we missed some cases.
If you can report a more complete backtrace maybe we can figure out whether what you propose is the right patch or we can find something better.

derkling, 2017-12-21:
The patch above should ensure we always have EM data in the platform data... please have a look and let me know if it helps on fixing your issues.

It it works, anyway, we should still capture the case where platform['nrg_model'] can be None and properly guard it too.

cdleonard, 2017-12-21:
I pushed my fixes on [this branch](https://github.com/cdleonard/lisa/tree/fix_no_nrg_model):
- https://github.com/cdleonard/lisa/commit/4bb42551f4241d2c76c0283a76a7a72e6906e011
- https://github.com/cdleonard/lisa/commit/fe1385ea940f3b5410850bbb650ea98db3c033a4

 Should I post two separate pull requests for this?

derkling, 2017-12-21:
> I pushed my fixes on [this branch](https://github.com/cdleonard/lisa/tree/fix_no_nrg_model):
> - https://github.com/cdleonard/lisa/commit/4bb42551f4241d2c76c0283a76a7a72e6906e011
> - [...]

They can be part of a single PR, thanks.
Just push a branch with both patches in and reference your issue
number in the PR message ;-)

> 
> https://github.com/ARM-software/lisa/issues/529#issuecomment-353357794

cdleonard, 2017-12-21:
I wrote a reduced variant for _load_em_from_device based on your code, pushed here: https://github.com/cdleonard/lisa/commit/c7227dc3a5135ca78ecdd4912cda6703fd2c5c38.

It's not clear how LISA configuration is supposed to work but one way would be for values to be picked in order of precedence from:
- the target_conf (which can be loaded from target.config on disk)
- the board conf (libs/utils/platforms/*.json)
- the device

Ideally all values could be picked from the device and the minimal config required would be stuff like ssh. Does this make sense?

derkling, 2017-12-22:
Yes that could work, thanks.

However, I would like to find give device device data precedence at least over the default board config. This can fix the case there a supported board (e.g. JUNO) gets a firmware update with some new DT data which change OPPs etc.

Can you move your call to `_load_em_from_device()` as part of `_load_em`?
Something as what I do in my last version  a00fab0

Then please update the commit message to reference the issue, by adding a text similar to the one I've used, and push your patch as a new PR which reference this one in the PR message... I'll give it a try and merge your version if it works for Travis and my setup.

cdleonard, 2018-01-03:
> However, I would like to find give device device data precedence at least over the default board config. This can fix the case there a supported board (e.g. JUNO) gets a firmware update with some new DT data which change OPPs etc.

Wouldn't such cases be better fixed by simply deleting stuff from board config? It makes sense to me to only add nrg_model manually when the data in dts is absent or somehow broken (as an override), but maybe I am misunderstanding this. The way this works is very confusing and perhaps "override order" should be explicitly documented.

Also, it would be nice if you could merge #532 separately because it makes sense as a standalone bugfix.

qperret, 2018-07-24:
Should be solved by #532

closed 2018-07-24 by qperret