ARM-software/lisa #533: Is it ok for EnergyModel_ClusterEnergy to measure non-monothonic cpu-busy-data?

cdleonard, 2017-12-21:
I tried to build an energy model with EnergyModel_ClusterEnergy.ipynb and I am getting data like this:

```
	cap	cluster	core	energy	freq
0	428	big	A72	-190.366532	600000
1	754	big	A72	-454.737333	1056000
2	976	big	A72	-5459.532065	1464000
3	1024	big	A72	-13245.334314	1596000
4	226	little	A53	-480.785152	600000
5	340	little	A53	-700.483116	900000
6	453	little	A53	-2525.761553	1200000

	cap	cluster	energy	freq
0	428	big	488.910997	600000
1	754	big	1045.501815	1056000
2	976	big	6753.157099	1464000
3	1024	big	15304.008077	1596000
4	226	little	587.965298	600000
5	340	little	878.468130	900000
6	453	little	3054.164654	1200000
```

The big cluster has non-monothonic per-cpu energy usage? This seems wrong and there are even some warnings about this however I tried to look into how this is calculated and found nothing obviously wrong. The specific warning is this:

```
WARNING : EnergyModel  : Active states powers are expected to be monotonically increasing.
```

It seems that most of the energy is somehow ascribed to the cluster rather than the individual cores and the per-core numbers are low and end up non-monothonic. The per-frequency energy and power graph looks good (power consumption is higher at higher frequencies) but the "per-cpu" slope is close to equal between those higher frequencies. Maybe the extra power usage from enabling an extra core really is close to the same at high freqs?

It could also point to something rotten in my platform related to PSCI implementation? In particular cpuidle seems to have no effect on power usage but this busy-cost-data seems to be based entirely on hotplugging to force CPUs off.

derkling, 2017-12-22:
I'll start from the last point: we have a series of "preliminary" tests which is worth to run in order to verify that the platform provides the basic framework working as expected for a proper EAS integration.
Have a run of:
   tests/eas/preliminary.py
   tests/eas/heavy_load.py

Regarding the energy model building notebook, you should consider that as a reference implementation which specifically (EnergyModel_ClusterEnergy ) is defined for systems where you can measure independently the energy consumed by each cluster. Is that your the case?

Another common culprit is that CGroups are used to isolate the CPUs to measure from the CPUs running the testing code. The freezer controller is also used to reduce the system noise.
These are all supports which is worth to verify are working as expected.

Finally, from you data above, if I understand correctly, all CPUs active energies are negative: that should not happen it appears to me like a possible issue in your measurement setup?

I cannot parse properly what you mean by "the "per-cpu" slope is close to equal between those higher frequencies."
Can you share a gist with your full notebook? maybe it makes it easy to read the data.
Sometimes you can get some interesting hints from the log reported while the notebook is executed...

cdleonard, 2017-12-22:
I do have power measurements on a per-cluster basis.

Here is the notebook in gist form (not the exact same run but shows the issue): https://gist.github.com/cdleonard/8909fe08e48db09b0b3961d6547d9b30

The P-state power numbers look credible to me even though the C-state numbers look broken.

As far as I can tell inside the pstates_model_df function it does a linear fit on energy usage by active core count. Then it estimates the energy for zero cores as the "cluster energy" and the per-cpu energy is the value for one core - cluster_energy.

This means the per-core energy is effectively the "slope" of the lines in the "cluster P-State profiling" graph. Those slopes are not monothonically steeper for higher freqs on my platform and I don't know if it's OK.

cdleonard, 2018-01-04:
One issue seems to be that the EnergyModel notebooks use the "userspace" governor while measuring idle power usage and the cpu could be a high freq. I'm not sure if that's the freq from the previous test (for P-states) or whatever it starts at when onlined.

At least for my platform idle power usage can vary with freq and the notebook ends up measuring WFI power usage at the maximum frequency. I get much idle lower power usage if I explicitly set minimum freqs in compute_idle_power. This could be considered a bug in lisa, potentially affecting other platforms as well. Right?

cdleonard, 2018-01-04:
I just realized my first question contains negative per-core energy numbers. I don't know how I got those but the question is not about them, just look at the gist instead: https://gist.github.com/cdleonard/8909fe08e48db09b0b3961d6547d9b30. The per-core energy numbers are positive but do not increase monothonically with increasing frequency.

cdleonard, 2018-01-04:
I am getting significantly more sensible idle power measurements with this simple patch inside compute_idle_power:

```
--- a/ipynb/energy/EnergyModel_ClusterEnergy.ipynb
+++ b/ipynb/energy/EnergyModel_ClusterEnergy.ipynb
@@ -840,6 +840,9 @@
     "                # Hotplug ON one more CPU\n",
     "                target.hotplug.online(cpu)\n",
     "                \n",
+    "                # Set lowest freq explicitly\n",
+    "                target.cpufreq.set_frequency(cpu, target.cpufreq.list_frequencies(cpu)[0])\n",
+    "                \n",
     "                cl_cpus = set(target.list_online_cpus()).intersection(set(cl.cpus))\n",
     "                logging.info('Cluster {:8} (Online CPUs : {})'\\\n",
     "                              .format(cl.name, list(cl_cpus)))\n",
```

It makes sense to me that when a CPU is onlined it starts at a high freq and WFI will consume more power than at a lower freq, but maybe I'm wrong. Perhaps I should open this as a separate issue?

douglas-raillard-arm, 2020-06-05:
Closing as the notebook has currently not been ported to current APIs. We can resurrect this ticket if needed

closed 2020-06-05 by douglas-raillard-arm