ARM-software/lisa #537: What is the best way to evaluate EAS versus non-EAS on Linux?

cdleonard, 2018-01-05:
After porting the EAS patches and I'd like to do a top-level comparison between EAS and non-EAS. Looking through lisa notebooks I didn't find anything that obviously fits; many of the tests refuse to even run without EAS.

If there is no good top-level synthetic evaluation for linux maybe I could use one for Android?

I'm looking for something that can say something like "EAS consumes x% less power".

derkling, 2018-01-09:
[ +eas_dev ]

Hi Leonard,
first of all I would like to inform you that, for EAS specific
question, you should better post your requests on the eas-dev ML:

   https://lists.linaro.org/mailman/listinfo/eas-dev
 
This is where you can reach most of the people working on EAS.

Hereafter are some comments related to your question.

On 05-Jan 06:41, Leonard Crestez wrote:
> After porting the EAS patches and I'd like to do a top-level
> comparison between EAS and non-EAS. Looking through lisa notebooks I
> didn't find anything that obviously fits; many of the tests refuse
> to even run without EAS.

Most of the tests we have on LISA_HOME/tests/eas are actually to test
that EAS is working as expected.

> If there is no good top-level synthetic evaluation for linux maybe I
> could use one for Android?
> 
> I'm looking for something that can say something like "EAS consumes
> x% less power".

We actually have a workflow to run a complete set of Android workloads
on a custom target and compare power/performance results corresponding
to different kernels. That suite is named wltest and it's part of
LISA.

Here are the instructions to run wltest:

   https://github.com/ARM-software/lisa/tree/master/tools/wltests

lemme know if you have any question/doubts about running that suite of
tests. Do please consider that the Google kernel team has a great
interest on checking wltests results to assess proposed scheduler
changes for the AOSP common kernel.

Here is an example of the report generated by wltest when comparing
WALT vs PELT kernels:

   https://gist.github.com/derkling/3a8c3568676a29e608d6dcb15af06241

As a final remark, please do notice that wltest currently supports
out-of-the-box only hikey960 boards with an ACME energy meter.
However, it's relatively easy to integrate the support for different
targets and energy meters. Unfortunately we do not have documentation
available, but... everything needed should be just what you find
under one of platform folders:

   https://github.com/ARM-software/lisa/tree/master/tools/wltests/platforms

You can just copy the content of:

   https://github.com/ARM-software/lisa/tree/master/tools/wltests/platforms/hikey960_android-4.4

and modify the contained files to match your specific target.

Internally we have an integration for Google Pixel 2 devices... but
still did not find time to push/merge it. Lemme know if you are
interested ;-)

Cheers Patrick

-- 
#include <best/regards.h>

Patrick Bellasi

cdleonard, 2018-01-09:
I was wondering if there way a way to generate good top-level power reports for plain Linux (not Android) using Lisa. It seems the answer is "no" so I'll close this question as answered.

I'll keep the android wltest stuff in mind though.

closed 2018-01-09 by cdleonard