ARM-software/lisa #582: benchmark test error

Kevin-WangTao, 2018-04-02:
I can run eas test properly with "lisa-test tests/eas/generic.py",
but when I try running jankbench test with "lisa-test tests/benchmarks/android_jankbench.py",
I got the error below,

nosetests: error: unrecognized arguments: -v --nocapture --nologcapture --logging-config=logging.conf tests/benchmarks/android_jankbench.py
2018-03-31 17:15:33,122 INFO    : root         : Using LISA logging configuration:
2018-03-31 17:15:33,122 INFO    : root         :   /home/kevin/tools/lisa/logging.conf
2018-03-31 17:15:33,122 INFO    : Benchmark    : === CommandLine parsing...
usage: nosetests [-h] [--boot-image BOOT_IMAGE] [--boot-timeout BOOT_TIMEOUT]
                 [--android-device ANDROID_DEVICE]
                 [--android-home ANDROID_HOME] [--results-dir RESULTS_DIR]
                 [--collect COLLECT] [--iterations ITERATIONS]
                 [--iterations-pause ITERATIONS_PAUSE] [--iterations-reboot]
                 [--iio-channel-map IIO_CHANNEL_MAP]
nosetests: error: unrecognized arguments: -v --nocapture --nologcapture --logging-config=logging.conf tests/benchmarks/android_jankbench.py
Failure: AssertionError () ... FAIL

I can find the description of the unrecognized arguments in what I got from "nosetests --help",
and it is quite different from the usage printed in the error print.

I can reproduce this issue with any benchmark tests in tests/benchmarks.

valschneider, 2018-04-06:
Hi,

Apologies for not replying sooner, I don't seem to be getting notifications when issues are raised.

The benchmarks are not supposed to be run with nosetest, so for jankbench issuing this on the command line would be enough:
```bash
./tests/benchmarks/android_jankbench.py
```

Sorry for the confusion, it doesn't help that the test folder isn't covered by the documentation (yet !).

FYI those benchmark files have somewhat been left to rot, so if you encounter issues with them the best would be to use WLtests (tools/wltests).

Kevin-WangTao, 2018-04-13:
thanks for reply, that works

closed 2018-04-13 by Kevin-WangTao