ARM-software/lisa #595: LISA test harness

joelagnel, 2018-04-23:
This issue is to develop a single test execution framework (test harness) to use by LISA:

Currently we have 3 ways:
* executor
* Android experiments
* WA

We should have a single streamlined way of running a workload.
Workloads can be any of the following:
- workload generator (wlgen)
- Android workloads (camera, uibench etc)
- Android behaviors (turn screen on/off)

joelagnel, 2018-04-24:
Yes sounds good, also probably that category can include wlgen stuff as well?

douglas-raillard-arm, 2018-04-24:
Most likely, unless there are strong advantages to define a "pure" wlgen workload.

valschneider, 2018-11-06:
I think that discussion can happen on the code itself now (see https://lisa-linux-integrated-system-analysis.readthedocs.io/en/next/kernel_tests.html#module-lisa.tests.kernel.test_bundle)

closed 2018-11-06 by valschneider