ARM-software/lisa #606: Cannot load EM from platform when some CPUs are hotplugged out - is it a problem?

credp, 2018-06-21:
My Juno board currently has CPUs 4 & 5 unplugged for some tests I want to run. When I try to load the energy model (from the sched_domains, i.e. DT-style full EM), the loader tries to load model data for CPUs which are not currently present in the sched domains. This is the log I get, which makes it pretty clear what happens, but I'm not sure if this is new or not.

I am using PR #605 right now.

2018-06-21 10:44:26,184 INFO    : TestEnv      : Attempting to read energy model from target
2018-06-21 10:44:26,824 INFO    : EMReader     : Attempting to load EM using from_sd_target
2018-06-21 10:44:28,421 ERROR   : TestEnv      : Couldn't read target energy model: No such file: '/proc/sys/kernel/sched_domain/cpu4/domain0/group0/energy/cap_states'
2018-06-21 10:44:28,423 INFO    : TestEnv      : Topology:
2018-06-21 10:44:28,424 INFO    : TestEnv      :    [[0, 3], [1, 2]]

credp, 2018-06-21:
The master branch can't load it either - same error is given. Question is, do we want to fix this and what does that really mean?

douglas-raillard-arm, 2018-06-21:
From the automated testing point of view, it is not a big issue since we load the EM once after boot, before anything happens on the target, so I'm assuming all CPUs are plugged in at this point.

douglas-raillard-arm, 2018-06-21:
Raising an exception may not be great, since the user could just want to load an EM corresponding to what EAS sees at this point in time, so I don't see any better way to handle that than logging the issue.