ARM-software/lisa #634: Lisa ipython startup issue with Vagrant VM

Fixfire, 2018-10-18:
Hello,

I'm experiencing issues with the command `lisa-ipython start` after the setup of the Vagrant VM.
In particular, after the command, the process exit in this way

`[1]+  Exit 127                nohup $cmd notebook --ip=$IPADDR --port=$PORT --NotebookApp.token=$TOKEN > $LOGFILE 2>&1  (wd: ~/lisa/ipynb)`

I read the `ipynb/server.log` and I found this error 

`nohup: failed to run command 'jupyter': No such file or directory`

In fact in the `/lisa/src/shell/lisa_shell` script if `iphyton` is not found the default command uses `jupyter`.
But both `which ipython` and `which jupyter` return empty sets.

So it seems that during the setup of the environment of the VM there is some problems with the installation of ipython or jupiter in the `install_base_ubuntu.sh`

I manually installed `ipython` and `ipython-notebook` packages and at the end the `lisa-ipython start` succeeded.
But... I cannot see any file inside the ipynb folder in the browser.

Thanks in advance for your help.

valschneider, 2018-10-30:
Hi @Fixfire, sorry for the delay, I wasn't getting notifications for issues (that should be fixed now).

> I manually installed ipython and ipython-notebook packages

Do you mean apt or pip packages? I think both exist and you can always get some misbehaviour by mix and matching versions.

Sorry for not being greatly helpful, cleaning up the vagrant configuration is on my todo-list, I'll try to get that done fairly soon.

Fixfire, 2018-11-06:
I've installed the apt packages, because during the  `install_base_ubuntu.sh` those packages are removed and the ones installed via pip are not found when I launch `lisa-ipython start` script

valschneider, 2018-11-06:
I've been playing with the Vagrant setup for our `next` branch. Could you give this branch a try? https://github.com/valschneider/lisa/tree/next-vagrant

It seems to work okay on my setup, but I can't seem to reach the jupyter server from my host - might be due to my local network setup.

Fixfire, 2018-11-06:
Ok, thanks.

I tried to setup the new VM with Vagrant from your branch `next-vagrant,` but during the `install_base_ubuntu.sh` script it faild with this error:

`default: ./install_base_ubuntu.sh: line 22: expect: command not found`

Fixfire, 2018-11-06:
Mine is the same `ubuntu/bionic64 (virtualbox, 20181105.1.0)`

Anyway, then all seems working, and now I can see the notebook from my host at localhost.

Now I'll try some tutorial examples with the HiKey960

valschneider, 2018-11-20:
See https://github.com/ARM-software/lisa/pull/648

PR #648 (2018-11-16): Vagrant update for lisa/next
Closes #634
closed 2018-11-20 by valschneider