ARM-software/lisa #635: tutorial: some notebooks are broken

v0lker, 2018-10-24:
https://github.com/ARM-software/lisa/pull/624
or maybe better to just delete the tutorial - if brokenness is not an issue they might as well not be there?

v0lker, 2018-10-24:
with lisa/next, there doesn't seem to be a LisaLogging class anymore, causing some more breakage.

Fixfire, 2018-11-06:
I encountered the same issues also in the `next-vagrant` branch at 
[https://github.com/valschneider/lisa/tree/next-vagrant](url)

v0lker, 2018-11-06:
i PRed fixes to just very few of the issues i encountered but it looks like the notebooks aren't going to be updated, but mostly removed -  see comment to the PR (#624)

valschneider, 2018-11-06:
@Fixfire On the `next` branch this one works: https://github.com/ARM-software/lisa/blob/next/ipynb/examples/kernel_tests/test_bundle_demo.ipynb. I'll try to find the time to do this notebook spring cleaning sooner rather than later, but in the meantime I'd recommend loading a trace in a notebook (if you execute one of those TestBundles you will have a trace loaded already) and just look at what the autocomplete suggests when typing e.g. `bundle.trace.analysis.`. I appreciate it's not the best of help, but we have to wear a hard hat right now...