ARM-software/lisa #702: ImportError: No module named 'wa_results_collector'

vingu-linaro, 2018-12-20:
I tried to run sched-evaluation-full.ipynb with next branch but face error:

---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-c8c2c71ef9f4> in <module>
      2 from IPython.display import display
      3 
----> 4 from wa_results_collector import WaResultsCollector
      5 import pandas as pd
      6 

ImportError: No module named 'wa_results_collector'

valschneider, 2018-12-20:
Hi Vincent,

wltests is probably the last block that hasn't been yet updated in lisa/next. It's still on my backlog, but there was an offline chat with @credp && @derkling suggesting that we could restructure it instead.

I merged https://github.com/ARM-software/lisa/pull/692 just today but I now realise I didn't move the wltests notebook into "deprecated". I'll see if I can just get it working instead.

vingu-linaro, 2018-12-20:
ok.
Our farm uses a python script based on sched-evaluation-full.ipynb (python2.7) to post process WAv3 results.
IIUC, you don't accept new changes in master branch so my plan was to move next branch and push our script (at least the .ipynb version).

valschneider, 2018-12-20:
I see, thanks for thinking about upstreaming it!

I'll have a look and see if I can unblock that shortly.

valschneider, 2018-12-20:
So looking into it it seems like `wa_results_collector` should mostly work, I think the issues I'm hitting are with the notebook code itself.

All imports in lisa/next have to be absolute, so to import `wa_results_collector` you should do:
```python
from lisa.wa_results_collector import WaResultsCollector
```
And see how that fares with your script. I'll keep looking at the notebook at my side and fix any issues I find.

valschneider, 2018-12-21:
https://github.com/ARM-software/lisa/pull/703 should get things working in next.