ARM-software/lisa #853: lisa/next: mkdtimg requires python 2

balsini, 2019-04-08:
As subject, this causes lisa-wltest-series on the next branch to fail for (at least) HiKey960.

valschneider, 2019-04-08:
I'm using ATF/UEFI on my HiKey960, which means using abootimg and that has worked fine for me (see https://github.com/ARM-software/lisa/pull/503). Sadly ATF/UEFI is still not the default since some variants of the board cause problems.

Let's see if we can get a python3-enabled mkdtimg...

balsini, 2019-04-08:
Since this is external code, maybe
```
-#!/usr/bin/env python
+#!/usr/bin/env python2
```
could be a solution to consider.

v0lker, 2019-04-08:
looks like you're lucky: https://github.com/96boards-hikey/tools-images-hikey960/commit/052b33813564e123da9aef8fcbacbd14d020f035

douglas-raillard-arm, 2019-04-09:
Using python2 for an external tool is not an issue, as long as they use the right `#!/usr/bin/env python2`  shebang as @balsini mentioned as per PEP 394 https://www.python.org/dev/peps/pep-0394/

douglas-raillard-arm, 2019-06-05:
Fixed

closed 2019-06-05 by douglas-raillard-arm