ARM-software/lisa #867: [Real scenes] Could LISA simulate by real scense?

billccchang, 2019-04-15:
Hi there,
I know LISA can simulate loading by rt-app, 
but is there any method can simulate real scense?
(I run LISA on android target, thanks!)

Thanks,
Bill

valschneider, 2019-04-15:
Hi Bill,

I suppose that by real scenes you mean Android apps?

If that is the case, https://github.com/ARM-software/workload-automation is what you're looking for - it is based on devlib (that's a target communication library) just like LISA, but focuses on reliably running Android benchmarks - Youtube, Geekbench, PCMark, Google Maps, and so on. You can record traces and then feed them to LISA for analysis.

We have a wrapper around workload-automation that also handles re-flashing your device for testing different commits of a given kernel (https://github.com/ARM-software/lisa/tree/next/tools/wltests), but depending on what you're doing it might be simpler to just use workload-automation as it is.

Hope that helps,
Valentin

douglas-raillard-arm, 2020-11-19:
Closing as no activity in a few months

closed 2020-11-19 by douglas-raillard-arm