ARM-software/lisa #1130: Couldn't find the sha1 of the kernel of the device

dlezcano, 2019-10-10:
Hi,
I've got the following error.

The kernel uname -r returns "5.3.0-qcomlt-arm64"

It was compiled with:
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- -j 12 Image.gz modules dtbs KERNELRELEASE=5.3.0-qcomlt-arm64

But LISA is complaining about not finding a sha1?
Thanks in advance

RuntimeError Traceback (most recent call last)
in
10
11 # Kernel tree used for the tests
---> 12 kernel_repo_path='/home/dlezcano/Linux/linux-qcomlt'
13 )

~/lisa/lisa/wa_results_collector.py in init(self, base_dir, wa_dirs, plat_info, kernel_repo_path, parse_traces, use_cached_trace_metrics, display_charts)
159 for wa_dir in wa_dirs:
160 logger.info("Reading wa_dir %s", wa_dir)
--> 161 df_list.append(self._read_wa_dir(wa_dir))
162 df = df.append(df_list)
163

~/lisa/lisa/wa_results_collector.py in _read_wa_dir(self, wa_dir)
382 # refactoring to use WA's own API for reading output directories.
383 df['_job_dir'] = df['id'].replace(job_dir_map)
--> 384 df.loc[:, 'kernel_sha1'] = self._wa_get_kernel_sha1(wa_dir)
385
386 return df

~/lisa/lisa/utils.py in wrapper(first, *args, **kwargs)
206 cache_map[first] = partial
207
--> 208 return partial(*args, **kwargs)
209
210 return wrapper

~/lisa/lisa/utils.py in partial(*args, **kwargs)
202 @apply_lru
203 def partial(*args, **kwargs):
--> 204 return f(ref(), *args, **kwargs)
205
206 cache_map[first] = partial

~/lisa/lisa/wa_results_collector.py in _wa_get_kernel_sha1(self, wa_dir)
591
592 raise RuntimeError("Couldn't find the sha1 of the kernel of the device "
--> 593 "that produced {}".format(wa_dir))
594
595 @memoized

RuntimeError: Couldn't find the sha1 of the kernel of the device that produced /var/www/wa_output/Linux/glmark2

valschneider, 2019-10-10:
Ah, seems like it really wants a sha1 and nothing else, so release versions stripped of sha1s will lead to this.

This should solve your problem:

``` diff
diff --git a/lisa/wa_results_collector.py b/lisa/wa_results_collector.py
old mode 100644
new mode 100755
index 99ff8bf4..0f6069eb
--- a/lisa/wa_results_collector.py
+++ b/lisa/wa_results_collector.py
@@ -578,9 +578,13 @@ class WaResultsCollector(Loggable):
             target_info = json.load(f)
 
         # Read the kernel release reported by the target
-        sha1 = KernelVersion(target_info['kernel_release']).sha1
-        if sha1:
-            return sha1
+        version = KernelVersion(target_info['kernel_release'])
+        if version.sha1:
+            return version.sha1
+
+        # No sha1, go for the release version
+        if version.release:
+            return version.release
 
         # Couldn't get the release sha1, default to reading it from the
         # directory name built by test_series
```

This will use `uname -r` when no sha1 is available - you'll have to make sure each of your builds have a different one, otherwise I'm fairly sure this is going to massively confuse (read, break in agonizing pain) wltest.

douglas-raillard-arm, 2019-10-10:
```
+        version = KernelVersion(target_info['kernel_release'])
+        if version.sha1:
+            return version.sha1
+
+        # No sha1, go for the release version
+        if version.release:
+            return version.release
```
can be simplified into:
```
+        version = KernelVersion(target_info['kernel_release'])
+        return version.sha1 or version.release
```

valschneider, 2019-10-10:
But that'll always return, right? I think if we don't have any release information we should still do whatever the rest of the code is doing (e.g. search for some release information elsewhere).

douglas-raillard-arm, 2019-10-10:
ah, indeed, the `if` was partial, there is an implicit `else` branch ...

dlezcano, 2019-10-13:
Hi,
thanks for the fix. I tried it but I've got a different error:

2019-10-12 12:40:20,188 INFO    : lisa.wa_results_collector.WaResultsCollector : Processing WA3 dirs matching [glmark2], rooted at /var/www/wa_output/Linux
2019-10-12 12:40:20,191 WARNING : lisa.wa_results_collector.WaResultsCollector : Trace parsing disabled
2019-10-12 12:40:20,194 INFO    : lisa.wa_results_collector.WaResultsCollector : Reading wa_dir /var/www/wa_output/Linux/glmark2
2019-10-12 12:40:20,207 INFO    : lisa.wa_results_collector.WaResultsCollector :  no data for /var/www/wa_output/Linux/glmark2/menu-glmark2-glmark2-1/poller.log
2019-10-12 12:40:20,211 INFO    : lisa.wa_results_collector.WaResultsCollector :  no data for /var/www/wa_output/Linux/glmark2/teo-glmark2-glmark2-1/poller.log
---------------------------------------------------------------------------
CalledProcessError                        Traceback (most recent call last)
<ipython-input-5-783494cfd519> in <module>
     10 
     11     # Kernel tree used for the tests
---> 12     kernel_repo_path='/home/dlezcano/Linux/linux-qcomlt'
     13 )

~/lisa/lisa/wa_results_collector.py in __init__(self, base_dir, wa_dirs, plat_info, kernel_repo_path, parse_traces, use_cached_trace_metrics, display_charts)
    165         for sha1 in df['kernel_sha1'].unique():
    166             if kernel_repo_path:
--> 167                 kernel_refs[sha1] = find_shortest_symref(kernel_repo_path, sha1) or sha1
    168             else:
    169                 kernel_refs[sha1] = sha1

~/lisa/lisa/git.py in find_shortest_symref(repo_path, sha1)
     41         "refs/heads/ refs/remotes/ refs/tags",
     42         universal_newlines=True,
---> 43         cwd=repo_path, shell=True)
     44     for line in branches.splitlines():
     45         try:

/usr/lib/python3.6/subprocess.py in check_output(timeout, *popenargs, **kwargs)
    354 
    355     return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
--> 356                **kwargs).stdout
    357 
    358 

/usr/lib/python3.6/subprocess.py in run(input, timeout, check, *popenargs, **kwargs)
    436         if check and retcode:
    437             raise CalledProcessError(retcode, process.args,
--> 438                                      output=stdout, stderr=stderr)
    439     return CompletedProcess(process.args, retcode, stdout, stderr)
    440 

CalledProcessError: Command 'git for-each-ref --sort=-committerdate --format='%(objectname:short) %(refname:short)' refs/heads/ refs/remotes/ refs/tags' returned non-zero exit status 128.

valschneider, 2019-10-14:
Hmph, looks like the whole thing relies on the `_wa_get_kernel_sha1()` returning an actual  git commit `sha1`... Can you change your build options to have the kernel version keep the commit ID of its HEAD rather than strip it out (i.e. let it be visible in `uname -r`)?

dlezcano, 2019-10-15:
Hi Valentin,
I reverted the patch and compiled the kernel with a commit id:
uname -r => 5.4.0-rc1-gf691ca8d72
In the wa_output/glmark2/__meta/target_info.json file, the kernel release is:
"kernel_release": "5.4.0-rc1-gf691ca8d72"

but I still have the error introduced by the patch above:
"CalledProcessError: Command 'git for-each-ref --sort=-committerdate --format='%(objectname:short) %(refname:short)' refs/heads/ refs/remotes/ refs/tags' returned non-zero exit status 128."

I restarted lisa, restarted the ipynb kernel and the cells but got the same error. Any idea?

douglas-raillard-arm, 2019-10-15:
Does the git command succeed when invoked in a terminal in kernel_repo_path ? If not, what does it complain about exactly ?

dlezcano, 2019-10-15:
Ah, good point. Indeed the command fail on this particular tree:
fatal: missing object 5365d2d0096878d32fb02dc2c2381a1597a3c2af for refs/heads/topology/4.21
Found it out finally, was a referenced linux tree renamed.

closed 2019-10-15 by dlezcano