ARM-software/lisa #1308: Object names in objects.inv files are not guaranteed unique; potential for incorrect doc URLs

bskinn, 2020-01-27:
**Describe the bug**

This is a ***potential*** bug: Note that the object names within an `objects.inv` are not guaranteed to be unique throughout the whole inventory. Thus, [this check](https://github.com/ARM-software/lisa/blob/38682ed28986d1bdb164b46f3c9ffa19e778b7bd/lisa/utils.py#L1358) is risky in some possible corner cases -- you might get a docs URL for an artifact other than the one you're looking for.

I'm pretty sure they're guaranteed to be unique within a code domain (e.g., `:py:`), though. If you're always looking for only Python objects, augmenting the conditional to something like `if inv_obj.name == obj_name and inv_obj.domain = "py":` should avoid any collisions. Otherwise, you might need to be more specific in your selection logic.

douglas-raillard-arm, 2020-01-28:
Hi @bskinn,

This issue is unlikely to actually materialize since it's only going to get URLs on our project, which only contains Python (at least the part that is documented with domain-specific directives). The fix is straightforward though so I'll do it.

Thanks,
Douglas

closed 2020-01-28 by douglas-raillard-arm

bskinn, 2020-01-28:
The collisions can also happen between, e.g., section headers (`:std:` domain) and modules (`:py:` domain). Completely agree it's a very low-probability case, but if it happens it'll be ~impossible to detect unless someone happens to notice the link going to the wrong place.

Hesitated to raise the issue given the low likelihood; would not have been offended at a "nah" response. 😃

PR #1309 (2020-01-28): lisa.utils: Make get_doc_url() robust against names of other domains
Inventory names might not be unique across domains so check for the py domain as
well:
https://github.com/ARM-software/lisa/issues/1308

#fixes #1308