ARM-software/lisa #1741: Fail to build kernel module 'sched_tp' with the mainline kernel

Leo-Yan, 2021-09-15:
**Describe the bug**
When I tried to build kernel module 'sched_tp' to enable ftrace events for scheduler, it complains:

**To Reproduce**
Steps to reproduce the behavior:
$ cd $lisa_folder
$ source init_env
$ cd tools/kmodules/
$ ./build_module ~/Work/opensource/linux/ ./sched_tp/

* what part of LISA is being used (synthetic tests, WA-related features, etc.)
tools/kmodules

* what kind of target is being used (linux, android, etc.)
Mainline kernel

* what code is being run if that is a custom script using LISA APIs
n/a

**Logs of the error**

```
[LISAShell kmodules] \> ./build_module ~/Work/opensource/linux/ ./sched_tp/
Building module for ARCH=arm64
make: Entering directory '/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp'
pahole -C file:///home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/vmlinux_deps.txt /home/leoy/Work/opensource/linux/vmlinux > vmlinux_deps.h
pahole -C file:///home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/vmlinux.txt /home/leoy/Work/opensource/linux/vmlinux > vmlinux.h
make -C /home/leoy/Work/opensource/linux M=/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp modules
make[1]: Entering directory '/home/leoy/Work/opensource/linux'
  CC [M]  /home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp.o
In file included from /home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h:10,
                 from /home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_events.h:18,
                 from /home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp.c:8:
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/vmlinux_deps.h:33:2: error: unknown type name ‘cpu_stop_fn_t’
   33 |  cpu_stop_fn_t              fn;                   /*    16     8 */
      |  ^~~~~~~~~~~~~
In file included from /home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_events.h:18,
                 from /home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp.c:8:
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h: In function ‘rq_of’:
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h:17:15: error: dereferencing pointer to incomplete type ‘struct cfs_rq’
   17 |  return cfs_rq->rq;
      |               ^~
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h: In function ‘cpu_of’:
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h:31:11: error: dereferencing pointer to incomplete type ‘struct rq’
   31 |  return rq->cpu;
      |           ^~
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h: In function ‘task_group_is_autogroup’:
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h:42:13: error: dereferencing pointer to incomplete type ‘struct task_group’
   42 |  return !!tg->autogroup;
      |             ^~
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h: In function ‘sched_tp_rd_span’:
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h:169:16: error: dereferencing pointer to incomplete type ‘struct root_domain’
  169 |  return rd ? rd->span : NULL;
      |                ^~
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h: In function ‘autogroup_path’:
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h:58:1: error: control reaches end of non-void function [-Werror=return-type]
   58 | }
      | ^
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h: In function ‘task_group_is_autogroup’:
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h:46:1: error: control reaches end of non-void function [-Werror=return-type]
   46 | }
      | ^
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h: In function ‘rq_of’:
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h:18:1: error: control reaches end of non-void function [-Werror=return-type]
   18 | }
      | ^
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h: In function ‘cpu_of’:
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h:35:1: error: control reaches end of non-void function [-Werror=return-type]
   35 | }
      | ^
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h: In function ‘sched_tp_cfs_rq_avg’:
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h:103:1: error: control reaches end of non-void function [-Werror=return-type]
  103 | }
      | ^
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h: In function ‘sched_tp_rq_nr_running’:
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h:178:1: error: control reaches end of non-void function [-Werror=return-type]
  178 | }
      | ^
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h: In function ‘sched_tp_rd_span’:
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h:173:1: error: control reaches end of non-void function [-Werror=return-type]
  173 | }
      | ^
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h: In function ‘sched_tp_rq_avg_irq’:
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h:148:1: error: control reaches end of non-void function [-Werror=return-type]
  148 | }
      | ^
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h: In function ‘sched_tp_rq_avg_dl’:
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h:139:1: error: control reaches end of non-void function [-Werror=return-type]
  139 | }
      | ^
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h: In function ‘sched_tp_rq_avg_rt’:
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h:130:1: error: control reaches end of non-void function [-Werror=return-type]
  130 | }
      | ^
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:273: /home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp.o] Error 1
make[1]: *** [Makefile:1847: /home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp] Error 2
make[1]: Leaving directory '/home/leoy/Work/opensource/linux'
make: *** [Makefile:18: build] Error 2
make: Leaving directory '/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp'
```

douglas-raillard-arm, 2021-09-15:
Hi @Leo-Yan , thanks for reporting this issue. I'm currently on holiday so I won't be able to push fixes but it seems that it's related to the header produced by pahole. What version of pahole do you use ? 

I don't remember exactly but the kernel requires a very recent version (for e.g. BTF) and I think we have similar requirements as older versions have various bugs. The pahole that ships on Ubuntu 20.04 is too old for kernel dev unfortunately.

If it turns out to be a pahole issue, i have short term plans to build the modules on the fly in a controlled environment with up to date tools so if you can live with a custom build for a couple extra months the problem should vanish.

The PR to track is https://github.com/ARM-software/lisa/pull/1730

Leo-Yan, 2021-09-16:
Hi @douglas-raillard-arm,

It's indeed the issue caused by pahole.  Ubunut's prebuilt pahole version is: v1.15; and I also built the latest pahole from scratch: v1.22, but it's no luck.

Simply to say, pahole fails to parse the data structure 'cfs_rq', 'rq' with the latest kernel source code.  I can reproduce the issue with the command:

```
$ pahole --show_reorg_steps --reorganize -C cfs_rq vmlinux
$ pahole --show_reorg_steps --reorganize -C rq vmlinux
```

It shows nothing for these two structures.

douglas-raillard-arm, 2021-09-17:
Just to confirm, was the vmlinux compiled with appropriate debug info ?

Leo-Yan, 2021-09-17:
On Fri, 17 Sept 2021 at 18:09, Douglas Raillard ***@***.***>
wrote:

> Just to confirm, was the vmlinux compiled with appropriate debug info ?
>

Yes, I confirmed the kernel was built with config  CONFIG_DEBUG_INFO.  So
the compile option '-g' has been enabled.

Leo-Yan, 2021-09-17:
Some update: this is caused by the kernel config: CONFIG_DEBUG_INFO_REDUCED.  If enable this option, some data structure debugging info will be removed from vmlinux.

So simply to say, we need to enable kernel config CONFIG_DEBUG_INFO and need to disable config CONFIG_DEBUG_INFO_REDUCED.

qais-yousef, 2021-09-17:
That's a limitation with pahole v1.15. I do have a workaround for that but it didn't make it LISA because I believe we'll have provide an environment to build the module as Douglas has mentioned which will hide these dependencies on pahole.

See here for example of a fix in Makefile:

https://github.com/qais-yousef/tracepoints-helpers/blob/pelt-tps-v3-create-events/sched_tp/Makefile#L26

Douglas, do you want to have this workaround in LISA or wait for the build env to come? I was under the impression we'll keep things clean for the latter.

douglas-raillard-arm, 2021-09-17:
From what I understood @Leo-Yan is already using 1.22 custom build following https://github.com/ARM-software/lisa/issues/1741#issuecomment-920697337

Leo-Yan, 2021-09-21:
Yes, I don't think this is an issue for me anymore,  eventually I think it's crucial for the headers generated by pahole with vmlinux.txt and vmlinux_deps.txt.

Just for your info, at my side, I used my self built pahole v1.22, I found the data structure is not ordered correctly, so even I used pahole v1.22, it still reports building failure:

```[LISAShell kmodules] \> ./build_module ~/Work/opensource/linux/ ./sched_tp/ 
Building module for ARCH=arm64
make: Entering directory '/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp'
make -C /home/leoy/Work/opensource/linux M=/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp modules
make[1]: Entering directory '/home/leoy/Work/opensource/linux'
  CC [M]  /home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp.o
In file included from /home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp_helpers.h:11,
                 from /home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_events.h:18,
                 from /home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp.c:8:
/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/vmlinux.h:40:29: error: field ‘cfs’ has incomplete type
   40 |  struct cfs_rq              cfs __attribute__((__aligned__(64))); /*   128   384 */
      |                             ^~~
make[2]: *** [scripts/Makefile.build:273: /home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp/sched_tp.o] Error 1
make[1]: *** [Makefile:1847: /home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp] Error 2
make[1]: Leaving directory '/home/leoy/Work/opensource/linux'
make: *** [Makefile:18: build] Error 2
make: Leaving directory '/home/leoy/Work2/Develop/tools/lisa/tools/kmodules/sched_tp'
```

I think the definition for data structure 'cfs_rq' should be placed very early so it can be used by other structures which use it.  In the other hand, pahole v1.15 can handle this correctly.  I uploaded the two headers generated by pahole v1.15 and pahole v1.22 respectively for your reference.

Please kindly close this ticket if you this is the issue for pahole rather than LISA.  Thanks for the info.
[vmlinux_deps_pahole_v1.15.h.txt](https://github.com/ARM-software/lisa/files/7199716/vmlinux_deps_pahole_v1.15.h.txt)
[vmlinux_deps_pahole_v1.22.h.txt](https://github.com/ARM-software/lisa/files/7199717/vmlinux_deps_pahole_v1.22.h.txt)

qais-yousef, 2021-09-22:
I did see ordering issues in the past. Some versions of parole did sort the list and others respected what we passed. That's one of the reasons I have vmlinux_deps.txt - I couldn't convince pahole to respect the order on some versions.

Can you confirm that cpu_stop_fn_t you had was on v1.15 please? I'll try to build v1.22 and see what error it generates and the best way to fix it.

Thanks!

Leo-Yan, 2021-09-23:
> Can you confirm that cpu_stop_fn_t you had was on v1.15 please?

Yes, I can confirm this with below commands:

```
leoy@leoy-ThinkPad-X240s:/tmp$ lsb_release -a
LSB Version:	core-11.1.0ubuntu2-noarch:printing-11.1.0ubuntu2-noarch:security-11.1.0ubuntu2-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal
leoy@leoy-ThinkPad-X240s:/tmp$ pahole --version
v1.15
```

qais-yousef, 2021-09-23:
Thanks for confirming Leo! I will check the new issue on 1.22 you reported and if there's a fix worthy pushing I'll do. I will try to bring up these inconsistencies with the pahole maintainer in hope there's something to be done about it upstream.

Leo-Yan, 2021-09-23:
Thanks a lot, @qais-yousef !

qais-yousef, 2021-09-25:
I couldn't reproduce your error with v1.22. It built fine for me. I wonder if there's a further config option that's causing the problem for you. If you want to share your .config with me I'd be happy to try it out to see if I can reproduce it. Thanks!

Leo-Yan, 2021-09-27:
Hi @qais-yousef,

Please see the enclosed kernel config file: 
[config_for_lisa.txt](https://github.com/ARM-software/lisa/files/7232888/config_for_lisa.txt)

And below is the info for my pahole building:

```
leoy@leoy-ThinkPad-X240s:~/Work/tools/pahole$ git log -1
commit 38fad22d669ab044314d015ddb3491d2dc6416fa (HEAD -> master, origin/tmp.master, origin/next, origin/master, origin/HEAD)
Author: Yonghong Song <yhs@fb.com>
Date:   Fri Sep 17 15:48:18 2021 -0700

    libbpf: Get latest libbpf
    
    Latest upstream LLVM now supports to emit btf_tag to dwarf ([1]) and the
    kernel support for btf_tag is also landed ([2]). Sync with latest libbpf
    which has btf_tag support. Next step will be to implement dwarf -> btf
    conversion for btf_tag.
    
     [1] https://reviews.llvm.org/D106621
     [2] https://lore.kernel.org/bpf/20210914223015.245546-1-yhs@fb.com
    
    Signed-off-by: Yonghong Song <yhs@fb.com>
    Cc: Alexei Starovoitov <ast@kernel.org>
    Cc: Andrii Nakryiko <andrii@kernel.org>
    Cc: Daniel Borkmann <daniel@iogearbox.net>
    Cc: bpf@vger.kernel.org
    Cc: dwarves@vger.kernel.org
    Cc: kernel-team@fb.com
    Signed-off-by: Arnaldo Carvalho de Melo <acme@redhat.com>
```

I built pahole with the steps:

```
1. install cmake
2. mkdir build
3. cd build
4. cmake -D__LIB=lib ..
```

douglas-raillard-arm, 2021-11-03:
@Leo-Yan I updated the PR for dynamic module compilation:
https://github.com/ARM-software/lisa/pull/1730
It now does not require being root anymore and uses overlayfs correctly (aka work dir on same FS as upper). The root permissions are avoided using user and mount namespace, similar to `unshare -rm` command.

douglas-raillard-arm, 2021-11-23:
Closing this issue as the issues seem to have been addressed and the `master` branch should now contain code that builds correctly (and mostly automatically). Please re-open an new issue if you face problems with the current state of LISA.

Since you tried the dynamic module compilation, the config might have changed slightly, see:
https://lisa-linux-integrated-system-analysis.readthedocs.io/en/master/setup.html#building-a-module

closed 2021-11-23 by douglas-raillard-arm