ARM-software/lisa #1742: Question: enable Arm energy probe (AEP) in ipython script

Leo-Yan, 2021-09-17:
I want to enable AEP in the ipython script, and then invoke the functions to reset meter and record the meter for a specific workload.  If I remember correctly, there have two interfaces for enabling AEP, one is the LISA's self interface and another is based on devlib.  So I am not clear what's a good practice to enable AEP in ipython script.

I searched a bit for the document, it just gives out the suggestion for using the below configurations:

```
# AEP Energy Meter configuration
aep-conf:
    # Channels to use
    # type: Mapping
    channel-map: ['Device0' : 'BAT']

    # Resistor values
    # type: TypedList[float]
    resistor-values: [0.033]

    # List of labels
    # type: TypedList[str]
    labels: ['aep']

    # TTY device
    # type: TypedList[str]
    device-entry: ['/dev/ttyACM0']
```

If this is the good way to enable AEP, then what's the function I should use in ipython script for energy measurement?  thanks for suggestions!

douglas-raillard-arm, 2021-09-17:
Hi @Leo-Yan ,

The current situation is:
* The code itself lives in devlib and is therefore shard with workload automation
* Lisa used to have implementation but they have been removed for all the bits that were just plain duplication of devlib 
* Lisa now only sticks to providing yaml configuration, which could be useful in a setting like with exekall, but I don't know if there is a real use case right now

If all you want is to instanciate a python object in a custom script, i would suggest using devlib directly (passing a Lisa target object should work), or using the wrapper class from Lisa if it has useful context managers etc (can't remember if it does). Unless you really need it, avoid the yaml conf, as all the keys are mapped to constructor parameters that you can invoke directly.

Once you have your instance, you should get devlib instrument API on it to get some data 
https://devlib.readthedocs.io/en/latest/instrumentation.html

That stack of APIs clearly needs some love and documentation, and there is also an idea floating around in devlib to unify the instrument and collector API. The future is probably in that direction.

Leo-Yan, 2021-09-21:
Hi @douglas-raillard-arm ,

Thanks for guidance.  Based on your suggestion, I read devlib and worked out the notebook as:
https://gist.github.com/Leo-Yan/df22336a035140d2207b63282a681e12

Seems to me, this is not too bad.  On the other hand, I'd like to suggest to add some examples in the repo so any user could easily enable the power meter; otherwise, the developers need to dive into the source code and find out what's they want.  But this seems like the scope of devlib and it causes extra workload for project maintenance, so it's up to your plan for this.

Anyway, thanks a lot for info and I am trying other features in LISA.   Appreciate your quick response on your holiday (sorry for disturbing and you could wait to reply until you come back!).

closed 2021-09-21 by Leo-Yan