ARM-software/lisa #1743: lisa-test command is broken

Leo-Yan, 2021-09-18:
**Describe the bug**
The command "lisa-test" is broken, when run it the tool reports failure "[2021-09-18 15:26:00,420][EXEKALL] ERROR  unsupported operand type(s) for |: 'type' and 'NoneType'".

**To Reproduce**
Steps to reproduce the behavior:
In the lisa shell, when execute the command  "lisa-test", it reports the failures.

**Logs of the error**
```
[LISAShell lisa] \> lisa-test 
exekall run lisa.tests --conf /home/leoy/Work2/Develop/tools/lisa/target_conf.yml --symlink-artifact-dir-to /home/leoy/Work2/Develop/tools/lisa/results_latest --share \*.Target --select-multiple 
[2021-09-18 15:26:00,420][EXEKALL] ERROR  unsupported operand type(s) for |: 'type' and 'NoneType'
```

Here have a question is: before there have configuration files eas/rfc.py and rfc.config, which can be used to run test case for enabling EAS and without EAS, then can compare the result.  This testing can be executed *automatically* by lisa-test.  I found the files eas/rfc.py and rfc.config have been removed from the repository.  Could you confirm what's the current status for this part?

douglas-raillard-arm, 2021-09-18:
Could you share the version of python you are using and your yaml configuration (with password redacted) ? A plain "lisa-test" proceeds with no issues on my side on python 3.8.

The eas/rfc.py and rfc.config have bit rotten for at least 4 years. Since end 2018 the automated test framework have been reworked. Nowadays, lisa-test is a shim around the exekall test runner. You can find some info in the online doc:
* Some kind of user guide on how to run them: https://lisa-linux-integrated-system-analysis.readthedocs.io/en/latest/workflows/automated_testing.html
* the API to write tests, and the documentation of the tests themselves https://lisa-linux-integrated-system-analysis.readthedocs.io/en/latest/kernel_tests.html

Note that you can also run the tests "manually" without using exekall in e.g. a notebook or a script as shown in https://lisa-linux-integrated-system-analysis.readthedocs.io/en/latest/workflows/ipynb/tests/synthetics_example.html

The only EAS specific tests are checking the task placement and are all subclasses of https://lisa-linux-integrated-system-analysis.readthedocs.io/en/latest/kernel_tests.html#eas-tests.

The rest is mostly to test PELT and some sanity in cpufreq behavior, or scheduler features like misfit.

douglas-raillard-arm, 2021-09-18:
Also, could could you report the output of this command on your setup ?
```
exekall --debug run lisa.tests --conf /home/leoy/Work2/Develop/tools/lisa/target_conf.yml --symlink-artifact-dir-to /home/leoy/Work2/Develop/tools/lisa/results_latest --share \*.Target --select-multiple 
```

This should provide the backtrack of the exception that you witnessed.

douglas-raillard-arm, 2021-09-18:
Another thing that is probably unrelated, but in the old days, installing Lisa used to do a nasty modification to .bashrc (or possibly another similar file like .profile) to add things to PYTHONPATH. This fortunately belongs to the past, but your .bashrc may still be scarred by that. If that's the case, please remove  this bit from your .bashrc.

douglas-raillard-arm, 2021-09-18:
I managed to reproduce, it looks like bokeh started using PEP604 style annotation in the release they made 3 days ago, before Python 3.10 is even out (and without flagging the package as requiring python >= 3.10). 
I'll implement the handling of these annotations when I'm back, meanwhile I suggest downgrading bokeh to 2.3.3.

douglas-raillard-arm, 2021-09-18:
I've opened a bug report on bokeh (https://github.com/bokeh/bokeh/issues/11637) and added a blanket fix for invalid annotations in exekall (https://github.com/ARM-software/lisa/pull/1744)

closed 2021-09-18 by douglas-raillard-arm