ARM-software/lisa #1747: Question: lisa-wltest-series on Hikey960 with Android

Leo-Yan, 2021-09-23:
Hi there,

I think now lisa-wltest-series is obsolete for Hikey960 with Android system.

The main reason is the command 'lisa-wltest-series' need to build the boot.img with different kernel commits, and comparing the performance and power data for these commits.  But you could see since now AOSP starts to support GKI kernel so that all the ko modules are required to be rebuilt when every time we build Linux kernel image, and we need to copy the Image and .ko modules into AOSP repository to generate the final boot.img and super.img.  From my understanding, when every time we update the boot image, we also need to flash super.img as well so can update all kernel modules (.ko files).

When I read the document tools/wltests/README.md, it gives information which is obsolete.  So could you give some suggestion for this part?  I personally think this is a very useful feature for EAS (or scheduler tunning).

Thanks!

douglas-raillard-arm, 2021-09-27:
Hi @Leo-Yan thanks for raising that,

wltest is definitely used inside our team as well. There has been some plans for years to evolve away from the shell script into a more robust Python solution, but no-one has acted on it yet (which is also why it has not been integrated in the online documentation, it's a bit of a relic, but one still in active use). I'll forward the issue to people that deal with android on a regular basis to see how we can fix it for GKI. If you happen to have something hacked on wltest already, don't hesitate to share it (even if it's kind of broken).

credp, 2021-09-29:
Hi Leo, it does not do that - we update the modules by remounting and overwriting them. It works fine for me with Pixel 4.

credp, 2021-09-29:
It might be possible (I haven't checked yet) to take the Floral definition and change a couple of the paths to make it work better. The H960 support is from Android 6 I think, so is very old nowadays.

credp, 2021-09-29:
OK, I looked into it a bit more. Our H960 support is definitely out of step with AOSP, but it was not intended to produce a coherent set of images to flash. That might have been the wrong choice at the time. The H960 support is unchanged as far as I can tell since 4.4-based kernels. What do you think is the right solution Leo?

Leo-Yan, 2022-03-12:
Since Android building is a time cost job, I usually use a build machine for building kernel image and Android images.   Just share the script I use for the building steps:

Below is the building script build_aosp_hikey960.sh I used on building machine:

```
#!/bin/bash

KERNEL_PATH=/home/leo.yan/repo-common-hikey960-5.10
AOSP_TOPDIR=/home/leo.yan/aosp/master

pushd $KERNEL_PATH
rm -rf out
BUILD_CONFIG=common/build.config.hikey960 build/build.sh
popd

pushd $AOSP_TOPDIR
. ./build/envsetup.sh
lunch hikey960-userdebug

rm $AOSP_TOPDIR/device/linaro/hikey-kernel/hikey960/5.10/*
cp -f $KERNEL_PATH/out/android12-5.10/dist/* \
        ${AOSP_TOPDIR}/device/linaro/hikey-kernel/hikey960/5.10/
cat $AOSP_TOPDIR/device/linaro/hikey-kernel/hikey960/5.10/Image.gz \
        $AOSP_TOPDIR/device/linaro/hikey-kernel/hikey960/5.10/hi3660-hikey960.dtb \
        > $AOSP_TOPDIR/device/linaro/hikey-kernel/hikey960/5.10/Image.gz-dtb
make TARGET_KERNEL_USE=5.10 HIKEY_USES_GKI=true -j24

popd
```

Then on my PC I can launch the building script remotely with ssh command:
```
$ ssh build_machine_ip /path/to/build_aosp_hikey960.sh
```

So I share the info at here.  If you think it's not urgent for support latest Android images for Hikey960, please feel free to close this ticket.  Thanks!

closed 2022-06-10 by Leo-Yan