ARM-software/lisa #1748: kind: host-fail run lisa

zf1575192187, 2021-09-26:
target-conf: kind: host

run demo :
from lisa.trace import Trace
from lisa.target import Target

target = Target.from_default_conf()

with Trace.from_target(target, events=['sched_switch', 'sched_wakeup']) as trace:
    target.execute('echo hello world')
    # DO NOT USE trace object inside the `with` statement

trace.ana.tasks.plot_tasks_total_residency(filepath='plot.png')

return err:
ValueError: The target's kernel needs CONFIG_FTRACE=y kconfig enabled

check the code and find
for path in ['/boot/config', '/boot/config-$(uname -r)']:
      try:
          return KernelConfig(self.execute('cat {}'.format(quote(path))))

/boot/config-$(uname -r)  $(uname -r) is not expanded, so don't find config，It's weird.

when I skip this bug and run it
the next bug is here:
Will not load cgroups devlib module: target is using systemd, which already uses cgroups
Cannot retrieve value of key nrg-model: Unable to probe for energy model on target.
Traceback (most recent call last):
  File "./zf.py", line 10, in <module>
    trace.ana.tasks.plot_tasks_total_residency(filepath='plot.png')
  File "/home/zhoufeng.zf/lisa/lisa/analysis/_proxy.py", line 59, in wrapper
    return x(**kwargs)
  File "/home/zhoufeng.zf/lisa/lisa/analysis/base.py", line 661, in wrapper
    hv_fig = f(**kwargs)
  File "/home/zhoufeng.zf/lisa/lisa/trace.py", line 4840, in wrapper
    return f(self, *args, **kwargs)
  File "/home/zhoufeng.zf/lisa/lisa/analysis/tasks.py", line 842, in plot_tasks_total_residency
    df = self.df_tasks_total_residency(tasks, ascending, count)
  File "/home/zhoufeng.zf/lisa/lisa/trace.py", line 4840, in wrapper
    return f(self, *args, **kwargs)
  File "/home/zhoufeng.zf/lisa/lisa/analysis/tasks.py", line 644, in df_tasks_total_residency
    for df in map(get_task_df, task_ids)
  File "/home/zhoufeng.zf/lisa/.lisa-venv-3.7/lib/python3.7/site-packages/pandas/util/_decorators.py", line 311, in wrapper
    return func(*args, **kwargs)
  File "/home/zhoufeng.zf/lisa/.lisa-venv-3.7/lib/python3.7/site-packages/pandas/core/reshape/concat.py", line 304, in concat
    sort=sort,
  File "/home/zhoufeng.zf/lisa/.lisa-venv-3.7/lib/python3.7/site-packages/pandas/core/reshape/concat.py", line 351, in __init__
    raise ValueError("No objects to concatenate")
ValueError: No objects to concatenate

I don't get it.

douglas-raillard-arm, 2021-09-27:
Hi @zf1575192187 ,

It looks like devlib is trying to read a file, with a path relying on shell expansion, but the path is quoted so the expansion is prevented. I'll make a PR to fix that. That said, it indicates that `zcat /proc/config.gz` failed since that is the fallback path, which either means:
1. you are using a distro kernel like the one from Ubuntu that does not enables /proc/config.gz (CONFIG_IKCONFIG_PROC=n)
2. You are missing `CONFIG_IKCONFIG_PROC=y` in your custom kernel config

```
 Will not load cgroups devlib module: target is using systemd, which already uses cgroups
```
This bit is expected if you use systemd, there is unfortunately nothing that can be done. systemd makes (legitimately) use of cgroups, and we cannot move tasks around without breaking service and slices management.

```
 Cannot retrieve value of key nrg-model: Unable to probe for energy model on target.
```
Nothing specific to worry about unless you are interested in EAS. In that case, it probably means that EAS is disabled on your target for one reason or another (missing config, missing energy model etc).

```
 ValueError: No objects to concatenate
```
Looking at the code, it seems to indicate that there is no task detected in your trace, i.e. `trace.task_ids` is empty. That is a bit strange, but at the same time if the kernel config was not loaded something may have gone wrong because of that when setting up ftrace. If problem persists once LISA can read the kernel config, we can have a look again at this error.

douglas-raillard-arm, 2021-09-27:
This should fix the quoting for the fallback path, which should fix the problem for distro kernels like Ubuntu's:
https://github.com/ARM-software/devlib/pull/572

zf1575192187, 2021-09-29:
> This should fix the quoting for the fallback path, which should fix the problem for distro kernels like Ubuntu's: [ARM-software/devlib#572](https://github.com/ARM-software/devlib/pull/572)

Ok, Thanks.

zf1575192187, 2021-09-29:
> Hi @zf1575192187 ,
>
> It looks like devlib is trying to read a file, with a path relying on shell expansion, but the path is quoted so the expansion is prevented. I'll make a PR to fix that. That [...]

I merge your fix, and LISA loaded config, but the error recurs, please take the time to take another look. Thanks.

[LISAShell lisa] \> python ./zf.py
Will not load cgroups devlib module: target is using systemd, which already uses cgroups
Cannot retrieve value of key nrg-model: Unable to probe for energy model on target.
Traceback (most recent call last):
  File "./zf.py", line 10, in <module>
    trace.ana.tasks.plot_tasks_total_residency(filepath='plot.png')
  File "/home/zhoufeng.zf/lisa/lisa/analysis/_proxy.py", line 59, in wrapper
    return x(**kwargs)
  File "/home/zhoufeng.zf/lisa/lisa/analysis/base.py", line 661, in wrapper
    hv_fig = f(**kwargs)
  File "/home/zhoufeng.zf/lisa/lisa/trace.py", line 4840, in wrapper
    return f(self, *args, **kwargs)
  File "/home/zhoufeng.zf/lisa/lisa/analysis/tasks.py", line 842, in plot_tasks_total_residency
    df = self.df_tasks_total_residency(tasks, ascending, count)
  File "/home/zhoufeng.zf/lisa/lisa/trace.py", line 4840, in wrapper
    return f(self, *args, **kwargs)
  File "/home/zhoufeng.zf/lisa/lisa/analysis/tasks.py", line 644, in df_tasks_total_residency
    for df in map(get_task_df, task_ids)
  File "/home/zhoufeng.zf/lisa/.lisa-venv-3.7/lib/python3.7/site-packages/pandas/util/_decorators.py", line 311, in wrapper
    return func(*args, **kwargs)
  File "/home/zhoufeng.zf/lisa/.lisa-venv-3.7/lib/python3.7/site-packages/pandas/core/reshape/concat.py", line 304, in concat
    sort=sort,
  File "/home/zhoufeng.zf/lisa/.lisa-venv-3.7/lib/python3.7/site-packages/pandas/core/reshape/concat.py", line 351, in __init__
    raise ValueError("No objects to concatenate")
ValueError: No objects to concatenate

Code is here
[LISAShell lisa] \> cat zf.py
from lisa.trace import Trace
from lisa.target import Target

target = Target.from_default_conf()

with Trace.from_target(target, events=['sched_switch', 'sched_wakeup']) as trace:
    target.execute('echo hello world')
    # DO NOT USE trace object inside the `with` statement

trace.ana.tasks.plot_tasks_total_residency(filepath='plot.png')

douglas-raillard-arm, 2021-09-29:
Can you try passing a path to `Target.from_target(..., filepath='trace.dat')` and then sharing the resulting trace.dat file ? The Trace object is supposed to load all specified events and then discard the trace file, but maybe it fails to load the events and then the file is gone.

Also you might want to add this in order to get a bit more info:
```
from lisa.utils import setup_logging
setup_logging()
```

note: you can put code and output between triple backquotes in order to preserve whitespace formatting in github markdown

zf1575192187, 2021-09-30:
> Can you try passing a path to `Target.from_target(..., filepath='trace.dat')` and then sharing the resulting trace.dat file ? The Trace object is supposed to load all specified events and then [...]

`cat zf.py
#!/usr/bin/python

from lisa.trace import Trace
from lisa.target import Target
from lisa.utils import setup_logging

setup_logging()
target = Target.from_default_conf()

with Trace.from_target(target, events=['sched_switch', 'sched_wakeup'], filepath='trace_test.dat') as trace:
    target.execute('echo hello world')
    # DO NOT USE trace object inside the `with` statement

trace.ana.tasks.plot_tasks_total_residency(filepath='plot.png')`

`[LISAShell lisa] \> python zf.py
2021-09-30 10:07:43,296 INFO    : root         : Using LISA logging configuration: /home/zhoufeng.zf/lisa/logging.conf
2021-09-30 10:07:43,302 INFO    : lisa.target.Target : Target configuration:
+- devlib:
    |- excluded-modules from default (list): []
    +- platform:
        |- class from default (str): devlib.platform.Platform
|- kind from user (str): host
|- lazy-platinfo from default (bool): False
|- name from default (str): <noname>
|- tools from default (list): []
+- wait-boot:
    |- enable from default (bool): True
    |- timeout from default (int): 10
2021-09-30 10:07:43,306 INFO    : lisa.target.Target : Creating result directory: /home/zhoufeng.zf/lisa/results/Target-<noname>-20210930_100743.306424
2021-09-30 10:07:43,307 INFO    : lisa.target.Target : User-defined platform information:

2021-09-30 10:07:43,403 INFO    : lisa.target.Target : Connected to target <noname>
2021-09-30 10:07:43,410 WARNING : lisa.target.Target : Will not load cgroups devlib module: target is using systemd, which already uses cgroups
2021-09-30 10:07:43,432 ERROR   : lisa.platforms.platinfo.PlatformInfo : Cannot retrieve value of key nrg-model: Unable to probe for energy model on target.
2021-09-30 10:07:43,500 INFO    : lisa.target.Target : Loading target devlib module cpufreq
2021-09-30 10:07:43,521 INFO    : lisa.target.Target : Loading target devlib module cpufreq
2021-09-30 10:07:43,541 INFO    : lisa.target.Target : Loading target devlib module sched
2021-09-30 10:07:43,641 INFO    : sched        : Scheduler sched_domain procfs entries found
2021-09-30 10:07:43,641 INFO    : sched        : Detected kernel compiled with SCHED_DEBUG=y
2021-09-30 10:07:43,641 INFO    : sched        : CPU capacity sysfs entries not found
2021-09-30 10:07:43,957 INFO    : lisa.target.Target : Effective platform information:
|- abi from target (str): x86_64
+- cpu-capacities:
    |- orig from target (dict): {0: 1024, 1: 1024, 2: 1024, 3: 1024, 4: 1024, 5: 1024, 6: 1024, 7: 1024}
    |- writeable from target (bool): False
    |- rtapp from target(platform-info/rtapp/calib),target(platform-info/cpu-capacities/orig),target(platform-info/cpu-capacities/writeable) (str): <depends on lazy keys: platform-info/rtapp/calib>
|- cpus-count from target (int): 8
+- kernel:
    |- config from target (TypedKernelConfig): <kernel config>
    |- symbols-address from target (DeferredValue): <symbols address>
    |- version from target (KernelVersion): 5.10.23.bsk.1-amd64 dirty SMP Debian 5.10.23.bsk.1-dirty Tue 30 Mar 2021 05:28:18 P
|- name from target-conf (str): <noname>
|- nrg-model from target (DeferredExcep): <lazy TargetStableError exception>
|- numa-nodes-count from target (int): 1
|- os from target (str): linux
+- rtapp:
    |- calib from target (DeferredValue): <lazy value of RTA.get_cpu_calibrations>
|- capacity-classes from target(platform-info/cpu-capacities/orig) (list): [[0, 1, 2, 3, 4, 5, 6, 7]]
2021-09-30 10:07:47,409 INFO    : lisa.platforms.platinfo.PlatformInfo : Attempting to read kallsyms from target
Traceback (most recent call last):
  File "zf.py", line 14, in <module>
    trace.ana.tasks.plot_tasks_total_residency(filepath='plot.png')
  File "/home/zhoufeng.zf/lisa/lisa/analysis/_proxy.py", line 59, in wrapper
    return x(**kwargs)
  File "/home/zhoufeng.zf/lisa/lisa/analysis/base.py", line 661, in wrapper
    hv_fig = f(**kwargs)
  File "/home/zhoufeng.zf/lisa/lisa/trace.py", line 4840, in wrapper
    return f(self, *args, **kwargs)
  File "/home/zhoufeng.zf/lisa/lisa/analysis/tasks.py", line 842, in plot_tasks_total_residency
    df = self.df_tasks_total_residency(tasks, ascending, count)
  File "/home/zhoufeng.zf/lisa/lisa/trace.py", line 4840, in wrapper
    return f(self, *args, **kwargs)
  File "/home/zhoufeng.zf/lisa/lisa/analysis/tasks.py", line 644, in df_tasks_total_residency
    for df in map(get_task_df, task_ids)
  File "/home/zhoufeng.zf/lisa/.lisa-venv-3.7/lib/python3.7/site-packages/pandas/util/_decorators.py", line 311, in wrapper
    return func(*args, **kwargs)
  File "/home/zhoufeng.zf/lisa/.lisa-venv-3.7/lib/python3.7/site-packages/pandas/core/reshape/concat.py", line 304, in concat
    sort=sort,
  File "/home/zhoufeng.zf/lisa/.lisa-venv-3.7/lib/python3.7/site-packages/pandas/core/reshape/concat.py", line 351, in __init__
    raise ValueError("No objects to concatenate")
ValueError: No objects to concatenate`

zf1575192187, 2021-09-30:
[trace_test.dat.zip](https://github.com/ARM-software/lisa/files/7256121/trace_test.dat.zip)

douglas-raillard-arm, 2021-09-30:
Ok turns out the method was incorrectly not reporting a missing event, and instead was just ignoring the task. This is fixed by https://github.com/ARM-software/lisa/pull/1752 , which should make it raise an appropriate exception. You will need to add `task_rename` event to the event list you collect and then it should work.

closed 2021-10-08 by zf1575192187